nightscout/AndroidAPS #3001: Discussion: Why open loop when DST ?

robertrub, 2023-10-29:
I'm on DASH and @jbt7rr has Medtrum. I had a 3h open loop with the summer/winter time change but he did not...

[image]

As AAPS knows IOB and COB at the moment of the change, the only thing that can change between 2AM and 3AM is the profile.

It would be good if, after the time change, AAPS sends new time to pump, then sends active profile to pump automatically. This way, we can keep closed loop for DASH too

jbr7rr, 2023-10-29:
This is a thing that would needs to be done in the pump driver. For dash I can see if I can do something however, in the coming weeks I'm tight on time.

robertrub, 2023-10-29:
```
however, in the coming weeks I'm tight on time.
```

No hurry, you have 6 months to finish it 🤣

vanelsberg, 2023-10-29:
> As AAPS knows IOB and COB at the moment of the change,
When timeshift is backwards, think the problem is that the database values get "duplicates" in time:
BG values, but also and oRef calculations on IOB and COB (decay times) become invalid....
So IMHO better stop loop for some time!

Btw: DASH pumptime is updated but needs users OK to do that. Think I would like to keep that this way?

robertrub, 2023-10-29:
```
needs users OK to do that. Think I would like to
```

Why validating the time by the user is important?

robertrub, 2023-10-29:
```
BG values, but also and oRef calculations on IOB and COB (database entries, decay times) become invalid....
So IMHO better stop loop for some time!
```

Yes, going back in time might be complicated. Need to think about how it can be solved.

At least, on _time move forward,_ it will be easier to solve.

The Medtrum pump did not open the loop yesterday night. So, it can be done 👍

vanelsberg, 2023-10-29:
> Why validating the time by the user is important?
User should know about DST change, check pump, confirm all is ok....

For DASH, this is a Pod alert status as we know (beep, refresh, ack) for other events (like Pod expiration or Pod status issues
Purpose of the alert is to inform user and let him/her confirm. This was by design or else Pod alerts could just as well be logged and handled automatically, without user intervention?

vanelsberg, 2023-10-29:
> Why validating the time by the user is important?
User should know about DST change, check pump, confirm all is ok....

For DASH, this is a Pod alert status as we know (beep, refresh, ack) for other events (like Pod expiration or Pod status issues
Purpose of the alert is inform user and let him/her confirm. All Pod alerts could be handled automatically,

robertrub, 2023-10-29:
```
User should know about DST change, check pump, confirm all is ok?
```

I don't know about other pumps but with DASH, what is there to check and confirm? What options do you have other than accepting the DST change ?

Ok, for informing the user about the DST change in x hours (this is already done actually) but when it is time to change, user doesn't really have any choice.

robertrub, 2023-10-29:
```
Purpose of the alert is to inform user and let him/her confirm. This is why Pod alerts are not handled automatically..
```

We are not talking about pod alerts. In pod alerts, user must do something (as in _Ok, I heard that pod will end in 2 hours_ or _Ok, low insulin reservoir_ ...)

In each alert, user must prepare an action to come (change pod). 

With DST, the user has nothing to do. If user has no options to choose or to do, why bugger him/her with a useless question?

robertrub, 2023-10-29:
_Btw: Even though I got the DASH Pod alert (time on Pod) to be confirmed, DASH continued to operate even when my confirmation was 5 hours later._

The problem is not the alert. It is the open loop for 3 hours.

AAPS can send a message telling that pump has been set to the DST change if you want. It is useless but why not.

You phone, computer, car etc don't show an alert telling you that it has set the new DST time.

vanelsberg, 2023-10-29:
> The problem is not the alert. It is the open loop for 3 hours.

IMHO
**As all Loop calculations are time based the only sensible and safe(!) way to handle DST or TZ changes is to STOP the loop.
Because of safety I strongly feel we should not change this!**

vanelsberg, 2023-10-29:
Only know EROS/DASH but in general I do not see DST as a problem. However I can imagine users being confused and concerned about this. They should not be?

* User should know about DST
* User may need to take appropriate actions (YMMV).

Makes no sense trying  to "automate" this at this mostly depends the type of change (DST or TZ travel) as wel as pump & profile specifics as described in documentation?

vanelsberg, 2023-10-29:
> The problem is not the alert. It is the open loop for 3 hours.

Not sure how much of a problem this is actually is? AAPS switches to standaard basals which should be ok in most cases. If it is not you will get into trouble anyway, even without DST ;-)

Not sure if AAPS will zero-temp when BG goes below target while in open loop?

vanelsberg, 2023-10-29:
>  why bugger him/her with a useless question?

I get your point here, and maybe you are right.
But personally I do not think it is useless? Confirming means the user tells the Pod to update the time on the Pod "_now_".

Yes, this could be automated. But as this is not a trivial change in the Pod's operational logic it could have unknown side effects? It switches the Pod's internal default basal rates, could triggering a secondary alert (expiry) or even a Pod fault? So implementing such would need be seen lightly.

I would prefer to be around when Pod's time is reprogrammed. I know - I'm a tech ;-)

closed 2024-12-15 by robertrub