nightscout/AndroidAPS #3002: [EOPATCH2] AAPS deactivates the time change loop

fjpezuela, 2023-10-29:
Hello,

In Spain, the time has changed this morning (from 3:00 to 2:00).

Users of the EOPATCH2 pump have received a notification stating: "Daylight saving time change less than 3 hours ago - Closed loop disabled".

For 2 hours, we have been with the loop disabled, only receiving the programmed temporary basals.

This has happened to users with master version 3.2.0.1 as well as dev users.
[AndroidAPS._2023-10-29_02-00-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/13197623/AndroidAPS._2023-10-29_02-00-00_.0.zip)

Later, everything started to work normally.

Here is an extract from the log, where you can see when the message appears:

03:02:42.273 [DefaultDispatcher-worker-5] D/APS: [LoopPlugin.invoke():229]: invoke from Calculation for app.aaps.core.interfaces.iob.InMemoryGlucoseValue@396971f
03:02:42.280 [RxCachedThreadScheduler-26] D/NOTIFICATION: [NotificationStore.add():61]: Notification received: Cambio al horario de verano hace menos de 3 horas - Lazo cerrado deshabilitado
03:02:42.280 [DefaultDispatcher-worker-5] D/CONSTRAINTS: [ConstraintObject.set():28]: Setting value true -> false (DST in last 3 hours.)[DstHelper]
03:02:42.282 [DefaultDispatcher-worker-5] D/CONSTRAINTS: [ConstraintObject.getReasons():89]: Limiting original value: true to false. Reason: DstHelper: DST in last 3 hours.
03:02:42.283 [DefaultDispatcher-worker-5] D/APS: [LoopPlugin.invoke():236]: LAZO DESACTIVADO POR RESTRICCIONES
DstHelper: DST in last 3 hours.
03:02:42.284 [RxCachedThreadScheduler-94] D/XDRIP: [XdripPlugin$onStart$3.accept():122]: UPL Start
03:02:42.284 [DefaultDispatcher-worker-5] D/APS: [LoopPlugin.invoke():456]: invoke end

Image of the period where the loop was disabled (I muted the notification).

[image]

[image]

Thanks!

Dorayaki-84, 2023-10-29:
Hi,
I attach screenshots of 2 more users.
Where we can see that due to the time change the closed loop stopped working for 2 hours.
[image]
[image]

parapenT1sta, 2023-10-29:
Check in Other Pumps. Is working as expected. No bug.

https://androidaps.readthedocs.io/en/latest/Usage/Timezone-traveling.html#time-adjustment-daylight-savings-time-dst

fjpezuela, 2023-10-29:
> Check in Other Pumps. Is working as expected. No bug.
> 
> https://androidaps.readthedocs.io/en/latest/Usage/Timezone-traveling.html#time-adjustment-daylight-savings-time-dst

Thank you. I had read that, but I thought it was referring to older pumps, as the other pumps do set the time correctly. So, does the Omnipod Dash do the same thing? There is no specific mention of these pumps.

Regards, Fran.

parapenT1sta, 2023-10-29:
Other is different from older. :) Dash has also the same behavior.

https://github.com/nightscout/AndroidAPS/issues/3001

vanelsberg, 2023-10-29:
Confirming: By design, no bug.

robertrub, 2023-10-29:
Well, the design is not good then :)

I can understand 1 hour without loop (especially when time goes back with double entries for BGs etc) but not 3 hours.

Even the 1h should be treated internally and stay on closed loop (as with some other pumps) when DST changes.

MilosKozak, 2023-10-29:
write better code ;)

robertrub, 2023-10-29:
Before starting to write code (which I don't know how to), I need to think over the logic of how to threat the double entries or the 1h "blank" in the data. Will think about it.

We have 6 months to find a nice solution ;)

Can't understand why some pumps don't need open loop for 3h...

MilosKozak, 2023-10-29:
the only pump with history which handles DST properly is Dana-i
For pumps without history it's only SW thing

closed 2023-10-30 by MilosKozak