nightscout/AndroidAPS #3006: [DEV] AAPSclient misses downloads

szantos, 2023-10-30:
AAPS 3.2.0.1-dev a909f8653a-2023.10.25 NSClientV3 + websockets
NS dev v15.0.2 44f70db
Time 2023-10-28 around 0-5h
[Log Export 10-28.txt](https://github.com/nightscout/AndroidAPS/files/13204242/Log.Export.10-28.txt)
Similar to #2748 #2932

Happened in the morning after turning on the AAPSclient device. Data is in NS, but TBR won't download. Missing data won't backfill, even if I force close AAPSclient, delete cache and restart NSClient.

[image]

[image]

MilosKozak, 2023-10-30:
and full sync?

szantos, 2023-10-30:
You've already suggested full sync at another issue. I don't know what it does, as it is not documented.

What does a full sync do?
Is it only downloading data from NS?
Or may upload, too?
Can it mess up anything (like the local DB of AAPS or AAPSClient, or even the DB behind NS)?

MilosKozak, 2023-10-30:
what do you use on main AAPS ? v1 or v3?

szantos, 2023-10-30:
v3 everywhere

MilosKozak, 2023-10-30:
on AAPS client 
- reset db
- select NSCv3 tab and from menu do full sync

szantos, 2023-10-30:
After full sync missing data is there

closed 2023-10-30 by MilosKozak