nightscout/AndroidAPS #3007: [DEV] AAPSclient blocked connectivity without reason

szantos, 2023-10-30:
AAPS 3.2.0.1-dev a909f8653a-2023.10.25 NSClientV3 + websockets
NS dev v15.0.2 44f70db
Time 2023-10-29 06:22:25
[Log Export 10-29.txt](https://github.com/nightscout/AndroidAPS/files/13204326/Log.Export.10-29.txt)

NSClient gave Blocked by connectivity options error without any reason, as every option is on, and there is no WIFI SSID restiction. Toggling all of them off and on, force closing the app and clearing cache has solves the problem.

[image]

[image]

MilosKozak, 2023-10-30:
you simply lost connection to internet

closed 2023-10-30 by MilosKozak

MilosKozak, 2023-10-30:
improved anyway here https://github.com/nightscout/AndroidAPS/commit/51e4a984974702b1652bfcabc4fe315234b66fda