nightscout/AndroidAPS #3009: [DynISF] Between 23:00 - 00:00 TDD stats only 6 days

olorinmaia, 2023-10-30:
I noticed that DynISF stop working between 23:00 and 00:00. Culprit seem to be that for some reason only 6 days of TDD are shown in AAPS Statistics and thus DynISF stop working.

Reproduced in latest dev and master.

Before 23:00
[image]
[image]

After 23:00
[image]
[image]

MilosKozak, 2023-10-31:
wasn't it at DST change day?

olorinmaia, 2023-10-31:
In Norway we adjusted time back 1 hour 29. October 03:00.

olorinmaia, 2023-10-31:
If it matters, same issue happens today. Both dev + master

jakobheyman, 2023-11-01:
I can confirm this issue in dev.
[image]

TheNeskik, 2023-11-02:
I confirm this bug in master too.
[AndroidAPS_LOG_1698965455785.log.zip](https://github.com/nightscout/AndroidAPS/files/13245500/AndroidAPS_LOG_1698965455785.log.zip)

szantos, 2023-11-03:
Same here with dev

MilosKozak, 2023-11-03:
https://github.com/nightscout/AndroidAPS/commit/47758b6b27c6e8562d598e5c63ba0ffde6593cb3

closed 2023-11-03 by MilosKozak

marjolijnvanekkendonk, 2023-11-07:
I can confirm in master 3.2.01. No daylight savings since I’ve started using 3.2 on October 30th and daylight savings happened on the 29th. 
[image]