nightscout/AndroidAPS #3024: Feature Request: New Automation Condition: Reccuring time range

yhonatanb20, 2023-11-03:
Hello everyone.
Currently, you can either set "2.11.23 8:00" (time), "mondays 8:00" (reccuring time) or "8:00-9:00" (time range) as automation conditions. I'd very much like being able to set a reccuring time range condition, e.g., "mondays 8:00-9:00".
The reason is very simple: many of us have a schedule that is varies between days but is similar between weeks. E.g., I get up later on weekends. So for my night-time automation, I'd like to set "Sundays-Thursdays 23:00-7:00" and "Fridays-Saturdays 23:00-9:00".

MilosKozak, 2023-11-03:
you can .... change OR to AND

yhonatanb20, 2023-11-03:
Sorry, this makes no sense to me. In the "reccuring time" condition there's an "At" rubric, so I can't set e.g., "Sunday-Monday **and** 0:00-7:00", only "Sunday-Monday At 0:00 **and** 0:00-7:00" which makes no sense because it'll be treagered only at 0:00 rather than from 0:00 to 7:00.

yhonatanb20, 2023-11-03:
@MilosKozak Please reopen issue