nightscout/AndroidAPS #3026: Feature Request: Setting 100% Profile with no Duration via Automation

yhonatanb20, 2023-11-03:
Currenlty, in order to make an automation which restores profile percentage to 100%, you should use the "change profile to" action. However, this may be problematic if you use different profiles for different days, since the automation can't know which profile should be used (see issues #3024 and #3025). Trying to make a "set profile percentage to 100%" action produces nonsense, since it requires to set a duration and besides requires the profile to be already 100% as a precondition.

MilosKozak, 2023-11-03:
what is the exact usecase for this?

yhonatanb20, 2023-11-03:
Background: I have a weekday_profile for weekdays and a weekend_profile for weekends
Automation A: when bg rises set 130% profile percentage
Automation B: when bg stops rising/begins falling, restore profile to 100%
Now notice that neither of the automations has any way of checking which profile am I currently using (weekday_profile or weekend_profile) (see #3024 and #3025), therefore automation B can't use an action such as "change profile to weekday_profile".

robertrub, 2023-11-05:
```
Automation A: when bg rises set 130% profile percentage
Automation B: when bg stops rising/begins falling, restore profile to 100%
```

That is not the most secure way to use automations. Better to set them for a duration of 10-15 minutes. If the triggering situation persists, then the same automation will kick in a second, third... time.

yhonatanb20, 2023-11-05:
Yes, I always do set a duration but still prefer to set an additional counter-automation

robertrub, 2023-11-05:
Your _set profile to 100%_ will NOT work. When you set a profile % change, automatically AAPS adds a corrolary looking if the profile is at 100%. If it is not100%, the _set profile to x%_ will not kick in.

[image]

yhonatanb20, 2023-11-05:
This is percisely my point... That's the problem I'm referring to - you can't set the profile back to 100% via automation.
If it weren't so I wouldn't have created this issue...