nightscout/AndroidAPS #3028: [Master/Dev] BG Check in AAPS doesn't show BG value AAPSClient > Treatments > User entry

olorinmaia, 2023-11-03:
AAPS & Client 3.2.0.1-dev
Build: f08f32c581-2023.11.03
Nightscout version: 15.0.2

In 3.1 adding a BG check in AAPS careportal it appeared in AAPSClient in overview graph, treatments > careportal and treatments > "user entry" where value was showed.

In 3.2 BG value in Treatments > "User entry" seem to have disappeared. Does not matter if NSClientV1 or V3.

Logs & Screenshots:

AAPS: 
[AndroidAPS_LOG_1699049346841.log.zip](https://github.com/nightscout/AndroidAPS/files/13255075/AndroidAPS_LOG_1699049346841.log.zip)

[image]
[image]

AAPSClient:
[AndroidAPS_LOG_1699049331268.log.zip](https://github.com/nightscout/AndroidAPS/files/13255076/AndroidAPS_LOG_1699049331268.log.zip)

[image]
[image]

MilosKozak, 2023-11-08:
UserEntries are local log of user actions. They are not transferred anywhere

olorinmaia, 2023-11-08:
I see. It was visible in 3.1 im pretty sure. Its just that we cant see bg value in AAPSClient anymore when entered in AAPS. Workaround for us is to type in value in notes field.

MilosKozak, 2023-11-08:
https://github.com/nightscout/AndroidAPS/commit/617d722edda0b09e54795648082449c66df37aa3

closed 2023-11-08 by MilosKozak