nightscout/AndroidAPS #3030: CI error in bolus calculations

robertrub, 2023-11-04:
I've never had this bug before. I started dinner, used the wizard to bolus and validated (not really checking the numbers). Then, when it started injecting, I saw over 7U to be given... My dinner is often less than 5U. I stopped immediately the injection. 

My Dev version

[image]

I had my COVID injection and BG numbers were stuck a bit high. I set several times profile at 160% for 15-20 minutes to lower the BGs before dinner.

[image]

But the last one was long before dinner.

[image]

This is my profile

[image]

And this is the calc results... The CI doesn't correspond to anything (not even 160% of my profile CI).

[image]

When I used the wizard a second time, all worked ok as usual.

And the logs. Look around 19:52

[image]

MilosKozak, 2023-11-05:
I didn't see any bug in logs. Always was calculated with correct IC, except result of Wizard.
The only possible reason I see is you selected another profile for calculation in dialog. Don't you have one with IC 4,3?

MilosKozak, 2023-11-05:
BolusWizard dialog is not covered by detailed logging to be able say what did you select

robertrub, 2023-11-05:
I just checked and none of my profiles use a CI of 4.3 🤔

If there is no info in the logs, nothing we can do. Those repetitive actions we do without really putting in our full attention, I can't see how but maybe I did something strange... 

I'll close this issue and keep an eye open when using the wizard.

closed 2023-11-05 by robertrub