nightscout/AndroidAPS #3031: 3.2.0.1 Increased connection Problems with Omnipod Dash

bionictom, 2023-11-05:
Ever since updating to 3.2 I'm having many more problems with connection to my Omnipod Dash.
Multiple times of days, i'll get errors, where AAPS isnt' sure if SMB were delivered or profile changes didn't work.
On top of that , I just had an error Pod-Fault: 095 ALARM_WIRE_TEST_OPEN_GROUND which I didnt have before.

I have been usind Dev-Version with DynISF before with no connection problems.

[AndroidAPS_LOG_1699178398619.log.zip](https://github.com/nightscout/AndroidAPS/files/13259240/AndroidAPS_LOG_1699178398619.log.zip)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13259241/AndroidAPS.log)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

jbr7rr, 2023-11-05:
Do you know which dev version you were using before?

bionictom, 2023-11-05:
The one before 3.2.1 I've used was Dev "D" if that makes sense.

robertrub, 2023-11-05:
```
 I just had an error Pod-Fault: 095 ALARM_WIRE_TEST_OPEN_GROUND which I didnt have before.
```

That alert has nothing to do with AAPS. It is pod breakdown.

With the last master and Dev, DASH connections are working really fast and good. Can you check your pod's validity dates ? Can you use another lot of pods to test ?

bionictom, 2023-11-05:
Thanks for the feedback on that error.

Unfortunately, all the pods I have are the same lot-number. And i won't get any other ones for the next months

robertrub, 2023-11-05:
And the expiry date is ???

bionictom, 2023-11-05:
> And the expiry date is ???

2024-12-07

robertrub, 2023-11-05:
Sorry, with the expiry out of the solution, no more threads to pull in order to guess what goes wrong with your system 🤔

robertrub, 2023-11-05:
Do you have other BT connected to phone (ear/headphones?) any new powerful 5G WiFi router/extender ?

For all those who have tested the modified 3.2 version DASH, it was always rock solid, always better than with the "old" version.

bionictom, 2023-11-05:
The other device connected via bluetooth is a smartband, but that's been like it before the update too. And no Wifi changes in my apartment either.

Thanks for looking into it. Maybe a weird combination of phone/os (lineageOS)? Maybe a software update will change anything? Gonna try that.

robertrub, 2023-11-06:
We didn't have any other OS to test the modified code. So that can be an explanation but I'm not really convinced.

Try disconnecting smartband first (easier than flashing the phone ;) )

robertrub, 2024-01-27:
Hi

Did you update to 3.2.0.3 and see if you still have the connection problems? If you have no more the connection problems, please close this issue.

bionictom, 2024-02-10:
I did update, but the problems persisted. I have now upgraded my phone (from Poco F3 on Lineage OS to Poxo X6 Pro on  stock OS, and the problems went away. So I guess it was a phone/lineage problem?

closed 2024-02-10 by bionictom