nightscout/AndroidAPS #3033: dynISF calculated value trend in graph

painbrain81, 2023-11-05:
it would be useful, to have an overall picture, to have the possibility of showing the trend of the calculated ISF in the AAPS graphs (like autosens % for example). this would also allow you to have an idea of ​​how it moves over time.

szantos, 2023-11-06:
Agreed.
DynISF is a function of SGV and a TDD-mix. Seeing the TDD-mix over time would help, too.

----

@painbrain81 
As a workaround you could use this service:
https://rafaeldelrey-nsview-nsview-9csanz.streamlit.app/
Wait some minutes for wakeup. Select suggested.BG and suggested.variable_sens or reason.ISF

olorinmaia, 2024-04-22:
@MilosKozak I think this one can be closed as variable sensitivity has been added as a graph option in dev 3.3

olorinmaia, 2026-09-23:
Closing as this has been implemented in master 3.4

closed 2026-09-23 by olorinmaia