nightscout/AndroidAPS #3039: After upgrading to 3.2.0.1 it is not possible to connect to a new pod (Omnipod Dash)

yakovmiller, 2023-11-06:
getting the following error
Generic error: java.security.InvalidKeyException: Banned public key: 0000000000000............
[image]
[image]
[image]

When I go back to my previous version (3.1.0.2) I manage to connect to the new pod and then again I upgrade to 3.2.0.1 and then everything works fine. But the connection itself with 3.2.0.1 failed
I tried resetting bluetooth, turning the phone off and on, reinstalling, clearing cache and storage, nothing helped. All I have left to do is go back to the previous version to connect and then go back to the new version.
Attached is the log in which the error occurred several times starting at 22:51:49.573
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13273534/AndroidAPS.log)

jbr7rr, 2023-11-07:
So I understand correctly that you can pair a pod with 3.1.0.x and then update to 3.2 and pod works but if you try to pair new pod it fails with this error?

Can you give me your phone type and android version?
Also maybe a log of successful pairing with 3.1.0.x would be helpful but not mandatory at this point

vanelsberg, 2023-11-07:
@jbr7rr / FYI: I'v experienced not a single problem activating DASH using latest 3.2 dev versions...
Could this be a faulty update i.e. wrong build key for new version?

Yash907127, 2023-11-07:
> @jbr7rr/ לידיעתך: לא נתקלתי בבעיה אחת בהפעלת DASH באמצעות גרסאות הפיתוח האחרונות של 3.2... האם זה יכול להיות עדכון שגוי, כלומר מפתח בנייה שגוי עבור גרסה חדשה?

How can I check what you say?

Yash907127, 2023-11-07:
> @jbr7rr/ יכול לידיעתך: לא נתקלתי בבעיה אחת בהפעלת DASH באמצעות גרסאות הפיתוח של 3.2... האם זה להיות עדכון שגוי, כלומר מפתח בנייה שגוי עבור גרסה חדשה?

How can I check it?

Yash907127, 2023-11-07:
> אז אני מבין נכון שאתה יכול לקשר פוד עם 3.1.0.x ואז לעדכן ל-3.2 והפוד עובד אבל אם אתה מנסה לקשר פוד חדש עם 3.2.x זה נכשל עם השגיאה הזו?
> 
> אתה יכול לתת לי את סוג הטלפון שלך ואת גרסת האנדרואיד? גם יומן של התאמה מוצלחת עם 3.1.0.x יעזור

You got it right
My phone is Xiaomi F21pro Android 11
The attached log should also show the link I created with version 3.1.0.3, then reverted to 3.2.0.1
The last failed connection attempt was at 22:58:35.645, then I went back to version 3.1.0.3, connected and upgraded again to version 3.2.0.1. It should appear in the attached diary near the said time

Yash907127, 2023-11-07:
> So I understand correctly that you can pair a pod with 3.1.0.x and then update to 3.2 and pod works but if you try to pair new pod with 3.2.x it fails with this error?
> 
> Can you give me your phone type and android version? Also a log of successful pairing with 3.1.0.x would be helpful

You got it right
My phone is Xiaomi F21pro Android 11
The attached log should also show the link I created with version 3.1.0.3, then reverted to 3.2.0.1
The last failed connection attempt was at 22:58:35.645, then I went back to version 3.1.0.3, connected and upgraded again to version 3.2.0.1. It should appear in the attached diary near the said time

Yash907127, 2023-11-07:
> @jbr7rr / FYI: I'v experienced not a single problem activating DASH using latest 3.2 dev versions... Could this be a faulty update i.e. wrong build key for new version?

How can I check it?

vanelsberg, 2023-11-07:
> How can I check it?

You probably did, but just to make sure:
Verify your new version 3.2 was build with the same keyfile used to build your previous version 3.1.0.3

- When _updating_ version 3.1.0.3 to 3.2 by starting the .apk it should say "update" and after updating all your settings should still be there and AAPS running without you needing to do any additional configuration.
- Verify there is only one single AAPS app installed on the phone.

Yash907127, 2023-11-07:
Yes
After updating to 3.2, there is no need to reconfigure, everything remains as it was in 3.1
And only one AAPS is installed on the phone

Yash907127, 2023-11-07:
> Also a log of successful pairing with 3.1.0.x would be helpful

I found in the log the return to 3.1.0.3
22:59:24.920 [main] D/CORE: [MainApp.onCreate():107]: Version: 3.1.0.3
Shortly after, the connection should appear

jbr7rr, 2023-11-09:
The concerning part of the code is hard to understand for me, but it looks like there goes something wrong with either a call to secure.random() or another java.secure related call. Maybe @avereha can also have a look ?

Ideally we would need a stracktrace (can be generated using AndroidStudio and your phone connected to USB), if you are willing to do that contact me on discord and I can walk you trough it.

millerachel, 2023-11-11:
i did your insructions.
i saved the log in adb mode (cmd line)
is the saved file enough for you? 
if not - please let me know how to output the logcat in a better way ,,,
the error message is around 2:50\2:51
[filename2.txt](https://github.com/nightscout/AndroidAPS/files/13326453/filename2.txt)

Yash907127, 2023-11-11:
I did your instructions
I saved the log in adb mode (cmd line)
Is the cache enough for you?
If not - please tell me how to get the logcat in a better way,,,
The error message is around 2:50\2:51
[filename2.txt](https://github.com/nightscout/AndroidAPS/files/13326462/filename2.txt)

millerachel, 2023-11-13:
does anyone has an idea how to help me?

MilosKozak, 2023-11-13:
as you are the only one so far I expect it's phone issue
I'd try reset to factory state

CeDoMain, 2023-11-17:
Do ypu have GPS enabled when entering the connect dialog? Try removing permissions and then gain them again to aaps. I have no issues.

mordechi-david, 2023-12-29:
To my lenders too it happened to attach a log
If there is a solution I would love
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13790228/AndroidAPS.log)
Also with F21

vanelsberg, 2023-12-29:
> My phone is Xiaomi F21pro Android 11

This phone is not on the list of tested phones, known to work. Also, compatibility problems were reported with some Xiaomi phones in the past (Poco, Redmi).  As the F12pro is not on the list of tested phones this is likely is a phone specific problem or an issue with latest OS update.

Did you check with other users running the F21pro

The fact that AAPS version 3.2 was built on latest OS platform libraries libraries which could explain new compatibility problems with the newer version of AAPS.

**You can try:**
1. When activating a new Pod, disconnect all (!) BT devices (except for BYODA)
2. Disable any backgrond application that might use BT, disable any antivirus apps.
3. Uninstall AAPS, restart the phone, then re-install AAPS
4. Factory reset the phone to make sure there are no OS inconsistencies
(YMMV)

If none of the above solves your problem:
I's suggest move to a different phone/brand that is known to work successfully with Omnipod DASH.

mordechi-david, 2023-12-31:
Hi, thanks for the detailed answer, I personally know a few dozen people with a Xiaomi F21 that works fine for them with Omnipod Dash even with the new version so it can certainly be on the list of compatible phones, (I can also point out that some models of other phones that I know took the version 3.1 takes between 30 seconds and a minute to connect to the pod every time since 3.2 connects in 3/4 seconds) but it's probably a specific operating system version that causes this problem, I tried resetting AAPS as well as removing bluetooth devices and it didn't help.

Yash907127, 2023-12-31:
Hi Mordechai
This is a problem with one of the running versions of F21 PRO
Each store sells this device with a different version
The solution is to burn a new version to the device
The Russian version and all based on it work without problems (if you know what I mean about the Russian version)
There are people who burn versions, contact me privately and I can refer you to them

mordechi-david, 2023-12-31:
I know and have already recognized this matter today for testing, the older Russian version does not only work newer, but you are indeed right that it is not a bug that should be handled because it is also repaired on the above device in the following versions so that this problem can be closed

closed 2024-01-01 by MilosKozak