nightscout/AndroidAPS #3040: Tidepool is not receiving profile basal, only temp basal

campbellgl, 2023-11-07:
3.2.0.1
Tidepool only receives temp basals. Profile basals show as 0.00 in Tidepool. In my logs from today, 2023-11-06 from 09:31 to 11:41 (16:31 to 18:41 GMT) my profile basal was running (0.85), but Tidepool shows zero basal.
[AndroidAPS._2023-11-05_17-00-43_.8.zip](https://github.com/nightscout/AndroidAPS/files/13274915/AndroidAPS._2023-11-05_17-00-43_.8.zip)
[AndroidAPS._2023-11-06_17-00-51_.0.zip](https://github.com/nightscout/AndroidAPS/files/13274916/AndroidAPS._2023-11-06_17-00-51_.0.zip)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13274917/AndroidAPS.log)

CeDoMain, 2023-11-17:
I can confirm this problem!

rmitchell7, 2023-12-22:
Same for me, does anyone know if there's a way I can get this working or who I can ask for help?

TWilhoWI, 2024-01-02:
Also conform the problem

ohjay73, 2024-01-02:
mine works fine using latest dev

Bricki85, 2024-03-26:
Hi, I have the same Problem. I am on Version 3.2.0.2. Switches to Pump (Dash) recently. Therefore now it would be nice if I would see my Basal rates correctly.

@campbellgl did you solve your Problem somehow?

campbellgl, 2024-03-27:
@Bricki85 I haven't tried in dev yet. Still not working in 3.2.0.4

Bricki85, 2024-03-27:
Thanks @ohjay73 and @campbellgl
Will then wait for 3.3.0. I am not yet brave enough to go full dev yet.

vanelsberg, 2024-03-28:
For me (latest dev) Tidepool only shows current ("automated") basal.
AFAIKT It does not support viewing the profile...

[image]

Bricki85, 2024-03-28:
I think you misunderstood me or the thread. What I see for you is a lot of zero TBRs and additional TBRs. Furthermore there are some gaps in our diagram where no blue line is. Rather few but some. At mine you can see it better:
[image]
There you see some blue lines with zero and then you see gaps where no blue line is. In those gaps my normal basal rate should be shown like for example 1.1U/h etc. But it seems Tidepool only shows the adjusted TBRs (deviating from the default basal rate in the profile).
I do not know your normal basal rate and if it is in the picture you uploaded. At least for me none of those are in there.

ohjay73, 2024-03-28:
tidepool will only put in what your device is actually doing. Therefore if
you don't have a basal all day because SMB's are being used to replace
basal, you'll see nothing in basal.. same if only TBR's are used. It doesnt
care what your profile is, only what is actually being reported by the
device.

Bricki85, 2024-03-28:
I have basal lets say for easier disussion 1,1U/h over the whole day. I see that I have zero TBRs and increased TBRs (those are in there) but all the gaps where there is no blue line there should be my default 1,1 basal rate. I see it in AAPS but not in Tidepool.

vanelsberg, 2024-03-28:
> ... where there is no blue line there should be my default 1,1 basal rate. I see it in AAPS but not in Tidepool.
You assumption is wrong: when on zero-TBR there is no insulin administred. Ckeck your pump activity?
Basal rate in profile is insuline "at rest", AAPS algo will adjust for 0 to max U/hr basal every 5 minutes .....

Bricki85, 2024-03-29:
> You assumption is wrong: when on zero-TBR there is no insulin administred. Ckeck your pump activity?
Basal rate in profile is insuline "at rest", AAPS algo will adjust for 0 to max U/hr basal every 5 minutes .....

If Zero TBR is set I get no Insulin. That is clear. And that AAPS adjusts the TBR is also fine.

I checked the export of my data from Tidepool and there is only as said the devivations from the default basal rate in there. Here is the corresponding Screenshot von AAPS. I would assume Tidepool and AAPS look similar.
[image]

So if you compare both screenshots you will find that the default basal rate where AAPS did not adjust anything is missing.

So from my underestanding AAPS does not report the basal rate which is inside the profile to the backend? Does it work with Nightscout?

Maybe to make it even more clear if I have 1,1U/h in my profile for the whole day and I would not have one adjustment like zero TBR etc. Then I would see nothing inside Tidepool. No blue line whatsoever. And therefore my numbers in Tidepool for Basal would be 0 for the whole day. Whereever AAPS would say 26,40 at a whole.

CeDoMain, 2024-04-03:
Hi guys. I understand what @Bricki85 means and I have the same issue. I have attached three pictures which show the same day. One from AAPS, Nightscout and Tidepool. For me its also clear that Tidepool doesnt show an active default Basal rate. It shows only the active non default rates. And for me and my doctor thats not good for comparing and analysing - e.g. something is missing which should obviously be there.

[image]
[image]
[image]

Bricki85, 2024-04-03:
Yeah here we can see it good, thanks for the images.
Nightscout actually does it correct somehow.
Question would be why. Does Nightscout get different data than Tidepool? Or does Nightscout just do more in the backend with the same data what Tidepool gets?
Maybe I dig into the code on the weekend if I find the time to see at least if the data upload is the same (I would assume it is. But you never know)

CeDoMain, 2024-04-04:
Thank you @vanelsberg for sharing your diagrams! Its interesting what you have shown. For me NS definetly has some problems displaying zero temps. In my diagram there are also problems.

But also in your tidepool diagram I can see the default basal error which the TO descibes. When you carefully look at 4'clock in your aaps diagram, you can see dark blue areas at the start and end of the tbr curve. I marked these in red in the image below. and these dark blue areas show default rates active which are also not present in your tidepool diagram.
[image]

vanelsberg, 2024-04-04:
> But also in your tidepool diagram I can see the "default basal error" ....

I think this is about graph scaling and sampling. Once should not give to much weight to de graph details.
Mind that the graphs are intended to show "activity". For details, check your pump activity (NS treatments reporting)

vanelsberg, 2024-04-30:
@janvitos @CeDoMain @campbellgl  Confirming: Also seeing this.

janvitos, 2024-04-30:
> So the diagrams with the missing sections in tidepool also occur at your side?

Yes, exactly the same issue. Normal basals are missing and only temp basals are recorded. You can see the large gaps in basal in the screenshot below.

[image]

Stealthwick, 2024-05-16:
I am also experiencing this issue. It caused some confusion with the diabetes specialist who was trying to work out why so little basal was being delivered - turned out it was just unaccounted for due to only temp rates being counted.

Would love to see this get some attention as we're working hard in my country for our approach to be accepted by healthcare professionals!

janvitos, 2024-05-16:
> It caused some confusion with the diabetes specialist who was trying to work out why so little basal was being delivered

This was my exact worry. My son's endo has access to his Tidepool and normally prints out the data of the last 2 weeks. But I'll have to use Nightscout and send them the reports myself now because of this bug. Hope it gets fixed in the near future!

Stealthwick, 2024-12-09:
This issue remains unaddressed as far as I know. Is anyone currently working on a solution here? Fixing this would be helpful in keeping our HCPs onside!

RApfelbeck, 2025-01-27:
I have the same problem with the current version 3.3.1.3. The normal basal rate is not displayed, only adjustments. The ticket has been around for a while. Is there anything I can do to help process the ticket?

[image]

RApfelbeck, 2025-01-27:
I took a look at the source of the TidepoolUploader. Here in UploadChunk only the getTemporaryBasalsStartingFromTimeToTime are fetched from the persistenceLayer and not merged with the normal values from the active profile. The currently visible result is therefore as expected according to the current source code. 

@MilosKozak According to Git history you were mainly active in this area, are you still active there? Would it be ok if we merge the data with the profile (in the history of the method I also see something that could go in this direction). If you could give me a few tips on the anchor classes, I can also make a suggestion. 

[image]

RApfelbeck, 2025-01-27:
*corrected*

I took a look at the source of the TidepoolUploader. Here the getTemporaryBasalsStartingFromTimeToTime are retrieved from the persistenceLayer in UploadChunk and linked to the profile in fromTemporaryBasals. However, the number of entries in the loop / result seems to be determined by the list of temporary entries. 

@MilosKozak According to the Git history you were mainly active in this area, are you still active? Would it be ok from your point of view if we take the basic set from the profile and then merge it with the temporary data in the loop? If you could give me a few tips on the anchor classes, I can also make a suggestion

[image]

vanelsberg, 2025-01-27:
> .. if we take the basic set from the profile and then merge it with the temporary data in the loop?

If I may jump in here?  I think it would be incorrect to assume that during any gaps in the TBR list the pump is running a basal rate from profile. Profile values are inly input for the algo are not used for setting basal rates.

Did you compare what is in Tidepool against what is logged under AAPS Treatments->Temp basal?
In my case (DASH) I only see few minor gabs, hardly noticeable. On close inspection in AAPS it looks like sometimes a TBR runs just a bit longer than the actual duration that was set according to the logged event.

[image]

I think the gabs we are seeing in Tidepool are from uploading the TBR events into Tidepool where it apparently is showing the TBR based on the duration as reported in the event data uploaded.

__Assuming TBR's sometimes run longer than the duration set_, a possible fix could be extending the end of a TBR event to the start of the next TBR event (=extending duration) in the list (assuming setting a new TBR was just delayed)_

@MilosKozak Interested in your insights in this.

vanelsberg, 2025-01-27:
As a follow up, question is what is causing these "gabs" in the list of TBT events:
- Temporary pump disconnects?
- AAPS to busy or low phone performance preventing AAPS to timely picking up TBR commands causing a TBR to run longer than the original duration intended?

Stealthwick, 2025-01-28:
I’ve run side by side comparisons with Nightscout and AAPS. Some of my
screenshots are above.

The gaps occur whenever the pump is running profile basal rate rather than
temp basal. You can verify this by suspending the loop and noting the
absence of basal Tidepool.

RApfelbeck, 2025-01-28:
yes, for tidepool I can confirm the behavior, only temporary basal values are transferred, the total basal insulin amount in the display also reflects this. It is particularly good to see in the openLoop with few basal rate adjustments, then long stretches of basal data are missing. 

[image]

RApfelbeck, 2025-01-28:
When I look at the overview in the app, the graphics are correct. The chats for the basal rate are divided into 4 series: base, temp, basalLine and absolute. The calculation of absulute is also based on base and if a temp is active, this is transferred to absulute instead of base. In my opinion, a similar approach would also be necessary here to fill the areas without temp with base.

What do you think?

[image]

CeDoMain, 2025-01-28:
@RApfelbeck That sounds like a good idea! I want to remark that Nightscout has also Problems with the Basalrate. There is also Data missing - although with a different scheme. Probably there are more bugs with the basal transfer. @RApfelbeck Is the code you have shown in the datachain before nightscout or only before tidepool?

RApfelbeck, 2025-01-28:
> [@RApfelbeck](https://github.com/RApfelbeck) Is the code you have shown in the datachain before nightscout or only before tidepool?

the source referenced in [comment-2616779013](https://github.com/nightscout/AndroidAPS/issues/3040#issuecomment-2616779013) only refers to the tidepool export.

the referenced source for the display in the overview of the app in [comment-2618994486](https://github.com/nightscout/AndroidAPS/issues/3040#issuecomment-2618994486) is generally valid and does not refer to an export

I have left/kept the relevant classes open as tabs in the screenshot. So you can quickly find the used references in android studio via class search with double shift

Kopsida, 2025-07-28:
Tidepool feedback

Kopsida, 2025-07-28:
This is the feedback I received from Tidepool regarding this issue:

##- Please type your reply above this line -##
Tidepool Logo
Your request (40997) has been updated. To add additional comments, you may reply to this email or use the link at the end of this email to view the ticket in our support portal.

	
Dave Cintron (Tidepool)

Jul 28, 2025, 09:48 CDT

Hi, thanks for your email and for reaching out to Tidepool Support.
 
Tidepool does not maintain the integration from AAPS to Tidepool. If data is not making it from your app, I would recommend communicating this issue to the developer(s) of the app and request they correct this problem.
 
You're not the first person to mention this and it's persisted for years, so I'm not sure whoever is maintaining the integration is interested in resolving it.
 
Sorry I can't be of more help. Let us know if you have any other questions.
Dave C.
Tidepool Support
support.tidepool.org
Tidepool Logo

	

Jul 28, 2025, 04:07 CDT

I upload my data from Android APS to Tidepool.

However, the reports in Tidepool don't show the basal profile, only the delivered or temporary basal amount.

The instructions on your website suggest that I should be able to see both values in the report,useful for ensuring that my basal rate are optimal.

Is there something I'm doing incorrectly?

Is be grateful for your help.

Link to this support ticket: https://support.tidepool.org/hc/requests/40997

[1RLP5K-LZ35D]
Many thanks for your prompt reply.
Show quoted text

CeDoMain, 2026-01-16:
I can also confirm that this isnt fixed in 3.4.0.0. :( What can we do that this get fixed soon? Any chance to get the attention of the devs who knows the tidepool export? I know that @MilosKozak recently developed the OAuth for Tidepool. Does he know how to fix this?

loudestnoise, 2026-04-01:
Hi, Dave from the Tidepool Support team here. I'm quoted in a comment here a while back 😂.

I'm not on our engineering team, but I live with diabetes and use an Open-Source automated insulin delivery (OS-AID) app too. I help support people with diabetes use the Tidepool platform and we have heard from a number of AAPS folk like y'all about this basal issue.

Does looking at the developer documentation help understand what's going on here? 

https://tidepool.redocly.app/docs/device-data/data-types/basal/scheduled

It sounds like Tidepool gets temporary basal rates ([automated](https://tidepool.redocly.app/docs/device-data/data-types/basal/automated)) but isn't getting the scheduled ones. I'm wondering if that's because the way that data doesn't line up with what Tidepool's data model expects.

The other widely used OS-AID apps on iOS use a [Tidepool Service](https://github.com/LoopKit/TidepoolService) to send data and this helps it conform to Tidepool's data model.

MilosKozak, 2026-10-10:
Fixed in AAPS 4.0 (since 4.0.0-beta1, 79227e4ea7).

When no temporary basal runs, AAPS now uploads the profile basal as "scheduled" basal segments, so Tidepool shows the profile basal line instead of 0. Closing.

closed 2026-10-10 by MilosKozak

janvitos, 2026-10-10:
> 30\. Fixed in AAPS 4.0 (since 4.0.0-beta1, [79227e4](https://github.com/nightscout/AndroidAPS/commit/79227e4ea71da24c9222bd2e1db1181216fdc557)).
>      When no temporary basal runs, AAPS now uploads the profile basal as "scheduled" basal segments, so Tidepool shows the profile basal line instead of 0. Closing.

Thanks!

When is AAPS 4.0 due for release on master @MilosKozak?