nightscout/AndroidAPS #3041: Can't backdate HW changes in Action Tab to yesterday

MowgliChild, 2023-11-07:
3103 and now 3201

CAN put today's date and other dates other than yesterday.

CAN'T put yesterday's date (jumps to day before yesterday)

Affects CGM, cánula, battery.

MowgliChild, 2023-11-08:
Hi Milos,
First up, great job.  Thanks 

I'm happy to add photos but dont see method to do so, just text editing. How do I add here?

I'm on S20FE so not a random phone.

It's changing the actual date, so not just a display issue, but totally accept you know way more than me.

May be easier just to try it your phone? Pick as HW, change insertion date to yesterday in calender (or typed in), click ok, screen goes back to text box input / confirm screen, but date changed.

Doesn't happen for today or any day other than yesterday.

MowgliChild, 2023-11-08:
[image]
[image]

MowgliChild, 2023-11-08:
Thanks for all your work Milos. Great job.

It's S20FE so not a random phone. 
Easy to replicate (on your phone). You don't need to commit changes.
Add via calendar or typing (yesterday is 7th), when you return to confirm screen the date is changed (6th). 
I don't see a work around. 
Issue only for yesterday, not any other day.
Input method seems identical 3201 and 3103, but honestly I'm not a programmer. I know nothing. 

Thanks again.

MilosKozak, 2023-11-08:
I works like expected.
Sorry I don't know

MowgliChild, 2023-11-08:
Super weird. Not a big problem. Feel free to close ticket since its just me then, haha
Thanks.

happygirlnyc70, 2023-11-21:
I can confirm that I am seeing this as well on Pixel 4a 5g. Dates are off by 24 when selecting in the calendar. My work around has been to select dates 24 hours ahead of desired date (e.g., for 20Nov2023 I select 21Nov2023)

Jared-Is-Coding, 2024-05-26:
I was just referred here from https://github.com/nightscout/AndroidAPS/issues/3264#issue-2183935650

Same issue.

MilosKozak, 2026-10-15:
test in v4 and open new issue if needed

closed 2026-10-15 by MilosKozak