nightscout/AndroidAPS #3043: hostorie smb AAPS (3.2.0.1) does not match historie Pump (Combo)

Gosee1960, 2023-11-07:
Sometimes i restarted my phone. So i did this evening around 20:04 or maybe 2/3 min earlier

After the restart i took my phone and my pum and was looking in the treatments (cob&smb part)
Also i looked in the historie of my pump. At that moment i saw a different in SMB

In AAPS there was a smb at 20:01 (0,5 eh) and at 20.07 a smb  (0,3) 
In my Pump there was a smb 20:01 (0,5 eh) and at 20:01 (0,3eh) and at 20:07 (0,3 eh)
So the difference is at 20:01 0,3 eh, i don't have clue what has happened. I didnt use any button on the pump and i am certain that i doesn't touch de array up or down.

So maybe there is a bug or there was something wrong after restarting ?

I hope there is a reason, thnax in advance

Greetz Jose

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html

- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

[Log Export.zip](https://github.com/nightscout/AndroidAPS/files/13289444/Log.Export.zip)
[AndroidAPS._2023-11-07_20-03-26_.16-JH.txt](https://github.com/nightscout/AndroidAPS/files/13289476/AndroidAPS._2023-11-07_20-03-26_.16-JH.txt)
[image]
[image]
[image]
[image]
[image]

MilosKozak, 2023-11-08:
what driver?

Gosee1960, 2023-11-09:
Combo v2 driverGreetz JoseVerzonden vanaf mijn Galaxy

dv1, 2023-11-16:
Unfortunately the logs do not show me what happened before 20:05. I suspect though that perhaps for some reason the Combo recorded the "standard bolus requested" event but not the "standard bolus infused" one. Impossible to tell for sure because of the missing logging, unfortunately.

In the maintenance section, press the gear icon to get to "Maintenance Settings". There, set the number of logs to 5 to be sure. Next time you observe this, _immediately_ send logs to be sure we catch the necessary details then.

dv1, 2023-11-17:
> Oke clear thanx foor your reply. I changed the number of logs to 5

And remember, produce the log as soon as you can if this happens again. I was unable to reproduce this, and it probably is a rare occurrence that requires a certain environment (for example, maybe something is special about your Combo, or your phone, some detail we are not aware of yet).

Gosee1960, 2023-11-17:
@dv1 I am the special in this case hahaha

MilosKozak, 2023-12-06:
closed for inactivity

closed 2023-12-06 by MilosKozak