nightscout/AndroidAPS #3044: Enhancement: add automation to switch DynISF Adjustment Factor %

RBJervey, 2023-11-08:
I'd love to see an automation to switch the DynISF Adjustment Factor % based on time of day. I'm sure others could see a use for this at other times too. Essentially it works great for me with this factor set to 100% all day long... But at 9pm its microboluses are suddenly crashing me. This is not a surprise. My ISF used to be set to 135 during the day and 198 at night time. I just need less insulin for corrections then. Ideally I'd be able to switch to an adjustment factor of about 60-70% from 9pm to about 3 or 4am. There doesn't seem any easy way to automate this change, I currently would need to go in and switch it every night at 9pm and then again at 3am. An automation to make this possible would be greatly beneficial.

RBJervey, 2023-11-08:
Actually... I think this is already being looked at in #2861. Sorry for not checking prior issues while it was crashing my blood sugar!

MilosKozak, 2023-11-08:
basically I don't agree to create a new algorithm on top of another algorithm on top of algorithm (Auto->DynISF->SMB)

szantos, 2023-11-09:
That's right.

Without DynISF a % profile change affected ISF, which can be automated. But with DynISF the only* way to affect the ISF is setting the DynISF factor - manually.

*not counting TDD-faking, which is surely not worth it

----

Obey/follow profile of #2767 would be a possible way to overcome this issue. 

----

Sidenotes:
1) % profile changes with DynISF are controversial. If the user wants to change the ISF along with the profile %, (s)he has to apply the same % to the DynISF factor, too.
2) Even more mistakes may happen, if the DynISF factor is already not = 100%, and has to be multiplied by another % (in head).
3) The 1d and 7d parts of the TDD-mix may catch up after some days, and the DynISF factor has to be touched again (while the profile % remains the same)...

MilosKozak, 2026-10-11:
Profile percentage now also scales Dynamic ISF, and automation can set a profile percentage for a time range. Please use that, for example for night settings. A separate automation action for the DynISF factor will not be added.

closed 2026-10-11 by MilosKozak