nightscout/AndroidAPS #3047: WearOS Client Feature Request: Local Vibration (or potentially audio as well) alerts for arbitrary BG levels.

PedanticAvenger, 2023-11-09:
Request: Would like to request the inclusion of vibe alert capabilities in the WearOS app triggered by configurable BG levels and that stay active until acknowledged.

Use Case:  I'm hard of hearing and audible alerts at night when I don't have hearing aids in  are not particularly effective at notifying me of low or impending low scenarios.  Being able to alert on say 4.1 mmol/L via vibe can wake me up to deal with before I really go low.

olorinmaia, 2023-11-09:
I guess you know, but you can use local broadcast of BG from AAPS to xDrip+ and in xDrip+ generate vibrate only alarms which can be sent to clock. You can also make automations in AAPS that generate alerts for vibrate only on clock if you like.

PedanticAvenger, 2023-11-09:
Yeah, I built several watch faces for fitbit that do just this.  The enhancement request is specifically to add that capability to the WearOS option so I can have both AAPS control (which you can't get via fitbit style integration) and alerts.

alex22rest, 2023-12-07:
I'd use the smart watch I bought a lot more if it has this feature. I went back to my fitbit because it buzzes easily when my glucose goes low or high, and I only use my smart watch with AAPS if I'm at a rare party where phones aren't allowed and might need to dose myself. Wasted hundreds of bucks on the smartwatch expecting the buzz feature