nightscout/AndroidAPS #3049: [Dev/Master] Automation icon missing and small bug with rule temp target equal like x mmol/l

olorinmaia, 2023-11-09:
1) If you add a automation rule with temp target "is equal to" example 8,0 it doesn't trigger. If you choose "equal or greater than" it triggers.
[image]
[image]

2) If you add a rule marked for "User action" the green gear icon doesn't appear in overview screen anymore, it was visible in 3.1.
[image]

MilosKozak, 2023-11-10:
@osodebailar i didn't find the reason why icon is not appearing. can you help?

MilosKozak, 2023-11-10:
https://github.com/nightscout/AndroidAPS/commit/27355c18a2acc217401b7d70624df95d0eadd918

ga-zelle, 2023-12-11:
With User Action the logo was always the gear icon. Therefore it does not add any information. Either reduce the button height or show the logo of the first action.

closed 2023-12-20 by MilosKozak