nightscout/AndroidAPS #3050: BG not rounded with dynISF

painbrain81, 2023-11-09:
After few days with dynisf i discover that in nightscout BG is "float" and not rounded. 
With dynisf off situation is normal with rounded values. 
I attach a pic (jusy shared in discord) with incorrect value. 
[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2023-11-11:
what is your BG source?

painbrain81, 2023-11-11:
It's always the same. Xdrip with freestyle 2 and smooth (by xdrip) with mean of last 5 values.
When dynisf is on, nightscout show float, when off is ok.
Ps: bg is uploaded by aaps, not xdrip

MilosKozak, 2023-11-11:
put here log from AAPS to be able see data coming from xDrip

painbrain81, 2023-11-12:
I'm sorry but I haven't been able to recreate the defect yet. I'll post the logs as soon as it happens

MilosKozak, 2023-11-21:
it's xdrip issue
i added rounding of inccoming values anyway https://github.com/nightscout/AndroidAPS/commit/3e681ef219ceb11b6a185548ef735c8e43bd97ea

closed 2023-11-21 by MilosKozak