nightscout/AndroidAPS #3051: Enhancement: Display DynISF sensitivity instead of Autosens on main screen

TheNeskik, 2023-11-10:
Now that DynISF can calculate a custom sensitivity. It could be great if this value can replace the autosens value since it is not used anymore with DynISF. It will be less confusing for people to see it on main screen instead of going to dynISF tab and looking for the calculated value.

MilosKozak, 2023-11-10:
it's just under AS value or?

TheNeskik, 2023-11-10:
We can see ISF value under autosens % you are right, but i'm talking about the option that we can turn on : "'Adjust sensitivity and basal" which calculate a sensibility using 24h TDD and 7D TDD. The result is different than autosens
[image]

olorinmaia, 2023-11-10:
maybe the icon should remove the "X" if that setting is turned on too, to illustrate that DynISF's TDD-generated sensitivity-ratio is active? 

The setting is further explained in wiki and dev now, and it would be clearer for user if icon and %-value reflect if setting is turned on or not

Staric17, 2024-04-06:
What exactly is the bonus if enabling "Adjust sensitivity and basal"?