nightscout/AndroidAPS #3054: Log Export

jchester47, 2023-11-11:
ADD TIME OF EVENT HERE:  9:08 PM
ADD ISSUE DESCRIPTION OR GITHUB ISSUE REFERENCE NUMBER: (Wear OS/Galaxy Wear): Tile commands do not work consistently when not using an AAPS supplier watchface.

If I use AAPS to send a warchface that comes with the app stock, tile commands work consistently. If using a different watch face, they only work for a while inconsistently and then eventually no longer get a response from the phone app.

Force closing the app and restarting it on the watch resumes function for a period of time before it quits working again.

I have tried all trouble shooting options I can such as ensuring enabling notifications and tuning off all battery restrictions on the watch. 

There seems to be a bug where the panel and the phone stop talking to each other after a certain period of inactivity when using a non AAPS watchface. However, complications continue working even when the panels no longer are. 
-------------------------------------------------------
(Please remember this will send only very recent logs.
If you want to provide logs for event older than a few hours,
you have to do it manually)
-------------------------------------------------------
AAPS 3.2.0.0
Build: 0d969ac8ca-2023.10.23
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: UNKNOWN
[image]

MilosKozak, 2023-11-11:
yes, android kills apps on background. Currently there is no solution

jchester47, 2023-11-11:
> yes, android kills apps on background. Currently there is no solution

This makes sense. Android OS battery optimizations are extremely frustrating as they misbehave in the background even when specified not to in the app settings. 

However, I do have one followup question: why does this issue not occur when using one of the AAPS supplied stock watchfaces? Do those force some kind of wakelock that prevents the app from sleeping or ceasing communication with the phone? And if the watch app is killed, how is it still supplying complication data and notifications?

MilosKozak, 2023-11-12:
Would be nice if you need a developer who can clarify ....

olorinmaia, 2023-11-13:
Two issues regarding same problem below, maybe we should link them and keep one issue for further investigating?

https://github.com/nightscout/AndroidAPS/issues/2336

https://github.com/nightscout/AndroidAPS/issues/2323

closed 2024-01-24 by MilosKozak