nightscout/AndroidAPS #3055: [Master]  Basal rate on AAPSClient doesn't update occasionally (V3+WS)

olorinmaia, 2023-11-11:
AAPSClient 3.2.0.1
NS 15.0.2

I've seen a few times today that openAPS-log says that temp 0 basalrate should be set, but it doesn't get updated in AAPSClient, I've checked AAPS to verify that a 0-basalrate has been set, and every time I've checked it has sucessfully set 0 basal but not getting updated in AAPSClient. I also see now in the Client screenshot that updated DyniSF-value is missing.

Logs:
AAPSClient: [AndroidAPS_LOG_1699740714936.log.zip](https://github.com/nightscout/AndroidAPS/files/13326754/AndroidAPS_LOG_1699740714936.log.zip)

AAPS:  [AndroidAPS_LOG_1699740741078.log.zip](https://github.com/nightscout/AndroidAPS/files/13326755/AndroidAPS_LOG_1699740741078.log.zip)

Screens
AAPSClient:
[image]

AAPS:
[image]

olorinmaia, 2023-11-11:
I havent' been able to reproduce in dev, alltho setup in my dev test-rig is random bg + virtual pump so not exactly the same, and I'm not monitoring it as much as my master-rig following my son :)

olorinmaia, 2023-11-12:
I'll update to 3.2.0.2 master and see if it still happens :)

olorinmaia, 2023-11-12:
Still happens in 3.2.0.2. On next BG basal gets updated and corrected in history.

AAPSClient / AAPS 3.2.0.2
Build: c31617fd2c-2023.11.12
NS: 15.0.2

Logs AAPSClient:
[AndroidAPS_LOG_1699812622823.log.zip](https://github.com/nightscout/AndroidAPS/files/13328891/AndroidAPS_LOG_1699812622823.log.zip)

Logs AAPS: 
[AndroidAPS_LOG_1699812649988.log.zip](https://github.com/nightscout/AndroidAPS/files/13328892/AndroidAPS_LOG_1699812649988.log.zip)

Screens
AAPSClient:
[image]

AAPS:
[image]

MilosKozak, 2023-11-21:
guessing time is not synced on both phones
code is ok

MilosKozak, 2023-11-28:
@olorinmaia did you verify?

olorinmaia, 2023-11-28:
Hi @MilosKozak  :) Sorry for late response. We got 3 phones following AAPS with NsclientV3+WS. All phones got standard android settings in regards of date and time. I can't really tell any time difference, if so it's milliseconds. I tried rebooting AAPS and my own phone also. 

What's odd is that it doesn't happen every time, and so far I've only noticed it when a temp 0 basal is issued or prolonged. I'm pretty sure it happens daily more than once but not so many times that this is something critical. It just make me worry some times that basal is still running when BG is falling alltho OAPS-log seem to suggest a temp 0 basal is set.

I can't remember this happening in 3.1.0.3 when using NSClient. I've not tested with NSClientV1 since I'm unable to change the 3rd AAPSClient NSClient-version as it at school at all times. (Is it a problem if an idle AAPSCLient is on V3 for a few hours while others are on V1?)

Screenshot of date/time settings:
[image]

olorinmaia, 2023-12-02:
Issue persists and happens a few times a day, it's not only temp 0 basal it's all basal updates. I'll bring another example if it can help improving this.

Notice at 20:19 OpenAPS-log suggest a temp 0 basal should be set 4 min ago. Basal is still showing 0,3 in AAPSClient. Nothing in NSCv3 log about temp basal update. Then at 20:20 temp basal is updated (back in time) for the temp basal being set by AAPS at 20:15.

AAPSClient 3.2.0.2
Build: c31617fd2c-2023.11.12
Logs:
[AndroidAPS_LOG_1701545151071.log.zip](https://github.com/nightscout/AndroidAPS/files/13537218/AndroidAPS_LOG_1701545151071.log.zip)

Screenshots AAPSClient.

### **20:19 (no temp basal enacted on AAPS 20:15 recieved in AAPClient yet)**

[image]
[image]
[image]

### **20:20 (temp 0 basal 120min enacted 20:15 finally recieved, just before new BG arrived).**

[image]
[image]

olorinmaia, 2023-12-03:
Switched to NSClientV1 yesterday, partly due to this issue and basal history is rendered wrong at Nightscout. This issue has yet to occur, so it seems to be related to NSV3 some how. :)