nightscout/AndroidAPS #3063: Different BG app and notification

painbrain81, 2023-11-15:
Hi.
I have smoothing active on aps.
In app i have a bg value but in notification i have another bg value... The same of bg source.
Is it correct?
I attach two pics done in the same moment.
[image]
[image]

jbr7rr, 2023-11-16:
Versions used? Which smoothing is used?

painbrain81, 2023-11-16:
Smoothing "in the middle" :)
Version 3.2.0.2

MilosKozak, 2023-11-21:
https://github.com/nightscout/AndroidAPS/commit/ef3a8fbffa15f6fab63938b7a8054fc0a283102d