nightscout/AndroidAPS #3065: [DEV] AAPSclient misses downloads

szantos, 2023-11-16:
AAPS 3.2.0.2-dev 2023.11.13 NSClientV3 + websockets
NS dev v15.0.3 5b440b7

Issue similar to #2748 #2932 #3006
Data is in NS, but won't download. Missing data won't backfill, even if I force close AAPSclient, delete cache and restart NSClient. Full sync would surely solve the problem.

----

Meal Bolus and Carbs missing
Time 2023-11-15 06:34
[Log Export 2023-11-15.txt](https://github.com/nightscout/AndroidAPS/files/13376447/Log.Export.2023-11-15.txt)

[image]

[image]

----

TBR missing
Time 2023-11-16 around 3-6h
[Log Export 2023-11-16.txt](https://github.com/nightscout/AndroidAPS/files/13376450/Log.Export.2023-11-16.txt)

[image]

----

Insulin Change also not synced.

szantos, 2023-11-21:
There is only 1 day out of 7, when no sync issues happen:

[image]

MilosKozak, 2023-11-21:
to be able address this issue I need short period of time when TBR was not synced (with logs from both sides)

szantos, 2023-11-21:
Thanks! If I can catch that, I'll upload the logs

szantos, 2023-11-22:
@MilosKozak 
Here you are - is that, what you've meant?

Missing in Client:

[image]

Data is there in NS:

[image]

Last and first synced SGVs (2023-11-21 20:21:13 & 2023-11-21 23:11:13) and TBRs in Client:

[image]

Logs:

[Log Export_Client.txt](https://github.com/nightscout/AndroidAPS/files/13437728/Log.Export_Client.txt)
[AndroidAPS_LOG_1700630344046.log.zip](https://github.com/nightscout/AndroidAPS/files/13437730/AndroidAPS_LOG_1700630344046.log.zip)

MilosKozak, 2023-11-22:
what is your NS connectivity settings?

szantos, 2023-11-22:
Both master and client every option on (no ssid restriction)

MilosKozak, 2023-11-22:
https://github.com/nightscout/AndroidAPS/commit/6de44f967548843c61490d7c2c811c321e2ff3ec

closed 2023-11-22 by MilosKozak