nightscout/AndroidAPS #3068: Issue with boluswizard Combo Combov2 driver

Gosee1960, 2023-11-17:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Time accident 09.43

With the boluswizard , 15 gr carbs for 40 min later, it says 1,4 bolus.
I dit OK
Then AAPS give a warning to me so i must check if bolus was given.
I did that, and bolus was given.
I checked in aaps the bolus/smb historie and also in the pump. They were both oke and the same

But the carbs in future was not registrated. 
I waited and check again but still no carbs in future, soa i manually at only 15 carbs.

I try to put the logs and some printscreens, maybe this will help

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13390066/AndroidAPS.log)
[image]
[image]
[image]
[image]
[image]
[image]

Thanx in advance

Greetz Jose

MilosKozak, 2023-12-06:
@dv1 the bolus failed (success = false reported to AAPS) thus carbs were not recorded.
But the bolus record was written to DB later.
can you check why?

blaqone, 2023-12-12:
Just had the exact **same problem** yesterday evening and I was confused - funny reading this now..

Added carbs manually then.

docx1, 2025-09-20:
I can confirm that this error is happening again. I thought that it was battery related, changed battery still the same error occurred.

MilosKozak, 2026-10-11:
Closing as a duplicate of #4240. The log here shows the same problem: the pump sends a reliable response a second time, the driver takes the copy as a new packet, the status read after the bolus fails, and the bolus is reported as failed (and the carbs are not stored) although the pump gave it. We continue in #4240.

closed 2026-10-11 by MilosKozak