nightscout/AndroidAPS #3070: Unwanted loop run when switching targets.

justmara, 2023-11-18:
When you switch targets this not happens atomicly. There is an additional loop run triggered - between target changes.

Why this is a problem?
Here is exact scenario that happened to me:

I have set up 'disable smb with high temptarget' for my daughter's aaps and we're successfully using it to prevent overreaction on sneaks, when countering hypos. Eat/drink  set 'hypo' target of 6.5 for 50 minutes and that usually is enough.

But this day after 35 minutes of  I decided to switch target to 6.0, so aaps could set a bit stronger TBS for the last 15 minutes. So without cancelling target I've set up NEW target of 6.0 for 15 minutes.
But AAPS did it in two steps: cancel TT, start new TT. Both events triggered loop run. So after cancelling TT AAPS placed a bolus for 'fast rise' that I wanted to prevent with high TT.

Lucky for us this time it was relatively small SMB, so we've managed it without a problem. But happens that predictions for the moment of change are highs so the unwanted SMB can be quite huge and lead to rollercoaster effect.

[image]

[image]

Target switch must be an atomic operation, without intermediate events and loop runs.

painbrain81, 2023-11-18:
It's true. Same for me. Between target switch aps launch for one moment loop and do smb if set.

swissalpine, 2023-11-18:
I can confirm this problematic behaviour.

MilosKozak, 2023-11-18:
well is not atomic. loop is triggered by database change and there are 2 changes: end current TT and start new TT.
I'll check the code if this an be improved

MilosKozak, 2023-11-21:
https://github.com/nightscout/AndroidAPS/commit/2feaa419a715000eff001e5d58ca12e073cca8f1

closed 2023-11-21 by MilosKozak

justmara, 2023-12-02:
got similar issue but with loop suspend event:
[image]

High TT were set, then loop were suspended, and enabled again later. On start in were triggered ignoring TempTarget so it placed SMB.