nightscout/AndroidAPS #3072: Log Export

jchester47, 2023-11-19:
ADD TIME OF EVENT HERE: Ongoing

ADD ISSUE DESCRIPTION OR GITHUB ISSUE REFERENCE NUMBER: 

I'm noticing an odd discrepancy on the latest version of AAPS when using exponential smoothing as recommended with the Dexcom G7. The glucose displayed within the AAPS app and on my watch reflect the smoothed value, however the glucose seen in the AAPS persistent notification in the notification tray shows the unaltered Dexcom reading. It's confusing as I cant see the calculated smoothed value without opening the app or checking my watch, which then causes me to worry my numbers are outdated when I see a discrepancy. Is anyone else experiencing this?

-------------------------------------------------------
(Please remember this will send only very recent logs.
If you want to provide logs for event older than a few hours,
you have to do it manually)
-------------------------------------------------------
AAPS 3.2.0.2
Build: c31617fd2c-2023.11.13
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: UNKNOWN
[image]

olorinmaia, 2023-11-19:
It's probably a bug, already reported here: https://github.com/nightscout/AndroidAPS/issues/3063

MilosKozak, 2023-11-21:
https://github.com/nightscout/AndroidAPS/commit/ef3a8fbffa15f6fab63938b7a8054fc0a283102d

closed 2023-11-21 by MilosKozak