nightscout/AndroidAPS #3074: 3.2.0.1/2 Loop Calculation Stopped Working

chrisbev, 2023-11-21:
3.2.0.1 loop calculation stopped working at just before 4:00 this morning.  Rebooted phone, asps, un/selected profile, pump, xDrip no luck. Samsung A41 Android 12 UI 4.1.  BG on main screen and xDrip tab but not on calculation tab. Graphs in History page.

[image]
[image]
[image]
[image]
[image]

[AndroidAPS._2023-11-21_00-00-03_.3.zip](https://github.com/nightscout/AndroidAPS/files/13425252/AndroidAPS._2023-11-21_00-00-03_.3.zip)
[AndroidAPS._2022-10-23_01-00-10_.1.zip](https://github.com/nightscout/AndroidAPS/files/13425253/AndroidAPS._2022-10-23_01-00-10_.1.zip)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2023-11-21:
logs are not current

chrisbev, 2023-11-21:
In case relevant Careportal pic added, the 3:45 may have been the start of it following we're probably be me to get things going.  I typically get a few a day as browsing when aaps restarts but otherwise seems OK. Ram at about 60% and storage about 90%.

chrisbev, 2023-11-21:
AndroidAPS.log attached hope that is what you need.

MilosKozak, 2023-11-21:
calculation worker never finish. This part is not covered by detailed logging.
Probably bad data in database. I'd reset database. If bad data is in NS too you'll need reset db + prevent loading it from NS

MilosKozak, 2023-11-21:
log ends at 3:19 and loop was still working

chrisbev, 2023-11-21:
Database reset from maintenance menu, profile set and still have problem.  Tried another profile and still have problem.  AndroidAPS(1).log attached in case it helps.

MilosKozak, 2023-11-21:
then pm me your config and master password

chrisbev, 2023-11-22:
I now have aaps loop calculating without uninstalling so will detaìl my actions/thoughts.
The treatments part of database had been reset and I was thinking about other parts of he db that could cause the problem and how to isolate which.  Stats has a separate reset and my AAPS App data is nearly 45MB even after clearing treatments so I thought there must be a lot of data.  Just in case I tried again setting different plugins from the config builder, see ...user.cvs attached in case order or which are important.  I then noticed Open Humans plugin was not selected and thought I had this running from ages ago!  Once I selected this (view was ticked and not altered) and then imported my preferences (over existing) to get back how I was the loop calculation started!  To try and repeat the problem I unticked OH and imported preferences and strangely what happened was the titles of the plugin tabs did not tie up from where OH should have been onwards and no last xDrip+bg data. 
Although not the original problem something a bit out of sync, I don't know how the code works and I fostered a postulation that if say the bg from xDrip goes to main screen and the xDrip tab but that the loop calculation is triggered by changes on the xDrip tab and some how the tab numbers (indexes) is one out and does not tie up with the tab that has xDrip data on a trigger will not then be triggered?
My problem possibly happened after an AAPS reboot (I get a few per day) so if say a minor race due to my h/w then I get to test quite a few over the years? Was it a glitch or: Are preferences saved correctly?  Are they imported correctly?  Is OH a red herring?
While looking I noticed that config builder > loop > APS mode was not set and if set for me didn't appear to do anthing is this by design or is a separate Issue required for this?

closed 2023-11-23 by MilosKozak

chrisbev, 2023-11-23:
Anyone here looking for similar problem, I was unable to recreate the original loop calc stopping which recovered after loading prefences (which had OH selected) on to aaps with OH selected and about to raise an issue for the incorrect loading of preferences when preferences had OH selected but aaps did not, as encountered above, when I thought it best to update to 3.2.0.2 first (this I had delayed due to watch update problems). For me 3.2.0.1 showed the incorrect plugin matching tabs title and data loading following an aaps restart but 3.2.0.2 was OK.  NB I get a few aaps restarts per day which I don't know if this is normal for this type of phone.