nightscout/AndroidAPS #3075: Add Tidepool to objective question

jbfeutz, 2023-11-21:
As Tidepool can now be used as an alternative to Nightscout this question should be updated to avoid confusion [image]

andyrozman, 2023-11-21:
Tidepool can't be used as alternative. Just as a backup.

MilosKozak, 2023-11-22:
tidepool is not prerequisity

closed 2023-11-22 by MilosKozak

jbfeutz, 2023-11-22:
I understand the comments above but this question is causing some confusion: https://www.facebook.com/groups/AndroidAPSUsers/permalink/3698473660374024/

If it is still the stance that Nightscout is a required prerequisite we should update the read the docs to reflect this as the new objective 1 makes it seem that it is no longer required.