nightscout/AndroidAPS #3077: Minimum SMB Interval is fix at 3 minutes

ga-zelle, 2023-11-23:
The user can select a time between 1 and 10 minutes as minimum SMB interval. Nevertheless  in LoopPlugin.kt for instance this minimum is hard coded as 3 minutes.

blaqone, 2023-11-23:
Interval set to 1 minute. This happens:

[image]

MilosKozak, 2023-11-24:
thanks. will be fixed in 3.3
https://github.com/nightscout/AndroidAPS/commit/37e9931255662c1abd4d8f4d51f5c9fdd0002eb4

blaqone, 2023-11-24:
Still does not work with your fix above included!

[image]

blaqone, 2023-11-25:
https://github.com/nightscout/AndroidAPS/commit/37e9931255662c1abd4d8f4d51f5c9fdd0002eb4#commitcomment-133429183

MilosKozak, 2023-11-26:
right. thanks!
https://github.com/nightscout/AndroidAPS/commit/dcf74d23e8b2fe8de51e416e307e33adcb8d1073

closed 2023-11-26 by MilosKozak

parapenT1sta, 2023-12-09:
@MilosKozak I left a comment on the commits. What is the reason of not being apply to "dev" branch? As I can see was applied on dev33, simplemode and equil branches.

parapenT1sta, 2023-12-09:
@MilosKozak What is the reason for this bug fix only be applied on the next 3.3 version?

MilosKozak, 2023-12-09:
3.3 diverged a lot
this fix is conflicting with changes in 3.3