nightscout/AndroidAPS #3081: Dash: cannot enter manual insulin injection without connected pump

cascer1, 2023-11-26:
When using the omnipod dash driver, and the pump isn't running (after occlusion, or between deactivating the old and activating a new pump, to name a few) the insulin button disappears.

The result of this is that I can't enter manual insulin injections after a pump error occurs in a place where I can't immediately replace it.

I remember some time ago (perhaps when I was using the Combo driver), the button still allowed me to enter manual injections.

I'd love for this functionality to be available again.

Using version 3.2.0.2

MilosKozak, 2023-11-26:
this is not possible because ability to enter insulin whe the pump is in error state is confusing for most people (they don't read "record only" note)
You can still do it afterwards or through NS

kenkolden, 2023-11-26:
i switch to Virtual pump/MDI  and enter insulin dose but mark  record only when i am off pump