nightscout/AndroidAPS #3083: AAPS 3.2.0.2 crashes on basal temp attempt -> No connection to ruffy...

b-aeby, 2023-11-26:
AAPS 3.2.0.2 (master) crashes every time a new basal temp is accepted.

Build version :  c31617fd2c-2023.11.21
Flavor: full release
Nightscout version 15.0.2

Pixel 6a, Android 14
Omnipod Dash

Here is the log file and a screen copy. It occurred at 16:16:28
16:16:28.996 [main] I/PUMP: [RuffyScripter.<init>():178]: No connection to ruffy. Pump control unavailable.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13467324/AndroidAPS.log)
[image]

vanelsberg, 2023-11-26:
Have not seen any of this. Assuming this might be phone specific?

b-aeby, 2023-11-26:
Objective 4, number of days > 7 but manual activation still at 0/20 as the app crashes on every attempts to accept a basal temp.

vanelsberg, 2023-11-29:
I'd say there is something in your profile or settings that AAPS can not handle...

1. Immediately after the crash, not date and time and then get recent logfiles and attach them to this issue so dev can have a look?
2. Unselect your current pump (set it to Virtual pump), close/restart AAPS (... menu, exit), reselect your pump.
3. To make sure this is not some invalid setting in your profile, try removing all profiles (make a note of current profile settings) and then recreate a new one.

jotomo, 2023-11-29:
@vanelsberg The log entry about RuffyScripter is benign. Will go away when ComboV2 has taken over.

b-aeby, 2024-01-04:
Hello, restarting setup from scratch, including all objectives solved this issue.
I didn't try to reproduce but the issue happened while objective 1 was passed with Tidepool only, no Nightscout.
Thank you

closed 2024-01-04 by b-aeby