nightscout/AndroidAPS #3088: Keep DynamicISF error reports insde the DynISF plugin

ga-zelle, 2023-11-29:
When DynamicISF detects insufficient TDD data or objective 11 not started it offloads the related reporting to the main SMB plugin. This confuses many users who are not on the DynISF trail. All such reporting should be contained and handled within the DynISF pugin before the fallback to main SMB plugin.

MilosKozak, 2023-11-29:
in dev there is notification added regarding fallback to SMB
What exactly do you mean? (screenshot)

ga-zelle, 2023-11-30:
Going through my archives I only found a screenshot showing the message about objective 11 not being started.  Earlier it used to be accompanied by messages about insuffucient history to calculate TDDs.
[image]
Because I never intend to use DynamicISF I commented out those commands in OpenAPSSMBPlugin.kt responsible for the tests I don't need. However, that is not a good idea for the general user. As  a test  I reactivated them:
```
        tdd1D = tddCalculator.averageTDD(tddCalculator.calculate(1, allowMissingDays = false))?.totalAmount
        tdd7D = tddCalculator.averageTDD(tddCalculator.calculate(7, allowMissingDays = false))?.totalAmount
        tddLast24H = tddCalculator.calculateDaily(-24, 0)?.totalAmount
        tddLast4H = tddCalculator.calculateDaily(-4, 0)?.totalAmount
        tddLast8to4H = tddCalculator.calculateDaily(-8, -4)?.totalAmount
        if (tdd1D == null || tdd7D == null || tddLast4H == null || tddLast8to4H == null || tddLast24H == null) {
            inputConstraints.copyReasons(
                ConstraintObject(false, aapsLogger).also {
                    it.set(false, rh.gs(R.string.fallback_smb_no_tdd), this)
                }
            )
            inputConstraints.copyReasons(
                ConstraintObject(false, aapsLogger).apply { set(true, "tdd1D=$tdd1D tdd7D=$tdd7D tddLast4H=$tddLast4H tddLast8to4H=$tddLast8to4H tddLast24H=$tddLast24H", this) }
            )
        }
```
The messages did not show up again because meanwhile I got sufficient history data. However, the TDD calculations are executed before each loop execution with time and battery power spent for nothing and should be done inside the DynamicISF plugin only. Here is an extract from the log after reactivating the test:

```
01:23:51.763 [DefaultDispatcher-worker-8] D/APS: [ProfilerImpl.log():16]: >>> SMB data gathering <<< executed in 322 milliseconds
01:23:51.894 [DefaultDispatcher-worker-8] D/CORE: [TddCalculatorImpl.calculate():116]: TotalDailyDose(id=0, version=0, dateCreated=-1, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1701212400000, utcOffset=3600000, basalAmount=4.981000000000002, bolusAmount=21.600000000000005, totalAmount=26.581000000000007, carbs=0.0)
01:23:51.895 [DefaultDispatcher-worker-8] D/CORE: [TddCalculatorImpl.calculate():67]: Storing TDD 1701212400000
01:23:52.137 [DefaultDispatcher-worker-8] D/CORE: [TddCalculatorImpl.calculate():116]: TotalDailyDose(id=0, version=0, dateCreated=-1, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1700694000000, utcOffset=3600000, basalAmount=5.643125000000001, bolusAmount=15.499999999999998, totalAmount=21.143124999999998, carbs=0.0)
01:23:52.363 [DefaultDispatcher-worker-8] D/CORE: [TddCalculatorImpl.calculate():116]: TotalDailyDose(id=0, version=0, dateCreated=-1, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1700780400000, utcOffset=3600000, basalAmount=5.591916666666668, bolusAmount=19.300000000000004, totalAmount=24.891916666666674, carbs=0.0)
01:23:52.549 [DefaultDispatcher-worker-8] D/CORE: [TddCalculatorImpl.calculate():116]: TotalDailyDose(id=0, version=0, dateCreated=-1, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1700866800000, utcOffset=3600000, basalAmount=6.155333333333334, bolusAmount=16.7, totalAmount=22.855333333333334, carbs=0.0)
01:23:52.728 [DefaultDispatcher-worker-8] D/CORE: [TddCalculatorImpl.calculate():116]: TotalDailyDose(id=0, version=0, dateCreated=-1, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1700953200000, utcOffset=3600000, basalAmount=5.554583333333337, bolusAmount=15.399999999999999, totalAmount=20.954583333333336, carbs=0.0)
01:23:52.911 [DefaultDispatcher-worker-8] D/CORE: [TddCalculatorImpl.calculate():116]: TotalDailyDose(id=0, version=0, dateCreated=-1, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1701039600000, utcOffset=3600000, basalAmount=5.022833333333336, bolusAmount=21.90000000000001, totalAmount=26.922833333333344, carbs=0.0)
01:23:53.049 [DefaultDispatcher-worker-8] D/CORE: [TddCalculatorImpl.calculate():116]: TotalDailyDose(id=0, version=0, dateCreated=-1, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1701126000000, utcOffset=3600000, basalAmount=4.551000000000002, bolusAmount=23.499999999999996, totalAmount=28.051, carbs=0.0)
[... cut ...] 000, utcOffset=3600000, basalAmount=1.236416666666667, bolusAmount=6.2, totalAmount=7.436416666666667, carbs=0.0)
01:23:53.150 [DefaultDispatcher-worker-8] D/CONSTRAINTS: [ConstraintObject.set():28]: Setting value true -> false (Autosens disabled in preferences)[OpenAPSSMB]
01:23:53.150 [DefaultDispatcher-worker-8] D/APS: [DetermineBasalAdapterSMBJS.invoke():80]: >>> Invoking determine_basal <<<
```
To me that looks like nearly 1.5sec wasted time and battery power for a setup without DynamicISF.

MilosKozak, 2023-12-04:
yes, so far won smaller code duplication vs wasting CPU time

ga-zelle, 2023-12-10:
I guess there is way to just pass on an error flag in case of fallback. Then just check the error flag in the SMB plugin instead of redoing the whole TDD calculation for a test. That way there is even no code duplication at all.

MilosKozak, 2026-10-11:
DynamicISF is now an option of the SMB plugin. The "fallback to SMB / no TDD" message is shown only when DynamicISF is enabled. Please reopen if you still see it with DynamicISF off.

closed 2026-10-11 by MilosKozak