nightscout/AndroidAPS #3089: Calibrations in Dexcom are causing duplicate value alerts and suspending closed loop

jchester47, 2023-11-29:
ADD TIME OF EVENT HERE:  11:22 AM 11/29/23
ADD ISSUE DESCRIPTION OR GITHUB ISSUE REFERENCE NUMBER:  Any time a calibration is entered in Dexcom, APS shows a red exclamation next to the blood sugar and a glucose value is marked as an out of normal sequence duplicate. However, this error occurs even when Dexcom does not accept the calibration and the Deccom app shows no duplicate or extra readings to match what APS shows in the history. Also, the supposed duplicate reading flagged in APS has a different time stamp from any other readings, confirming that it is not a duplicate.

This is very problematic as a calibration should not break the loop. I have to remember each time to go back into APS, delete the supposed duplicate value, and restart the loop. 
-------------------------------------------------------
(Please remember this will send only very recent logs.
If you want to provide logs for event older than a few hours,
you have to do it manually)
-------------------------------------------------------
AAPS 3.2.0.2
Build: c31617fd2c-2023.11.13
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: UNKNOWN
[image]

jchester47, 2023-11-29:
I am using the G7 app, however this issue occurred with the G6 as well.

MilosKozak, 2023-12-04:
* log doesn't have 11:22
* keep logging at default level until asked to change it

blaqone, 2023-12-05:
diaKem sending doube values to AAPS is a known problem.

MilosKozak, 2024-02-11:
Closed for inactivity

closed 2024-02-11 by MilosKozak