nightscout/AndroidAPS #3108: Short delta "15m" less than most recent delta of 5m

alex22rest, 2023-12-12:
The short delta "15m" is less than most recent delta of 5m even when every 5 minutes the blood glucose has risen. Long delta is also wrong. For example if BYODA is: 132, 129, 128, 127, then delta reads +3, 15m reads: +2, and 40m reads: +1. If the time is 306pm and the last reading was at 302pm of 132, and 40 minutes before at 222pm it was 119, and it gradually rose each time, then the 40m is +1. 
I've been told that it's not really measuring 15min and 40min, but even so, it is wrong.
I've had my screenshots criticized so I'll let the person reading this look for themselves - your 15m isn't your 15m delta, and your 40m isn't your 40min delta, and it isn't even a shorter timespan longer than your immediate delta.
It is useful to have it right for things like OpenAPS SMB Advanced Settings where "Always use short delta instead of simple delta: Useful when data from unfiltered sources like xDrip+ gets noisy". Thank you.
Build version: 3.1.0.3

MilosKozak, 2023-12-12:
long deltas are calculates as an average value of deltas in the past

alex22rest, 2023-12-12:
Thank you for your fast response! If I look at my BYODA, then if it is an average of the past 40m, it is data of: 132+129+128+127+127+126+124+122+119. That is not +1 like it says as 40m delta.

MilosKozak, 2023-12-12:
yes

MilosKozak, 2023-12-12:
meaning is not how much did bg change but what was the average change

alex22rest, 2023-12-12:
Thank you for clarifying. I'd asked other people and read the docs and didn't get a clear answer, but this answers it.

closed 2023-12-13 by MilosKozak