nightscout/AndroidAPS #3111: [dev33] ComboV2 - Pump site change as SMB

parapenT1sta, 2023-12-14:
I am using latest dev33 and my pump site change are being recognized as SMB (bolus)! I made a test to give 9.5U on a pump site change. Please check the screenshots and logs. Check around 8:00AM

[AndroidAPS_LOG_1702541012663.log.zip](https://github.com/nightscout/AndroidAPS/files/13670461/AndroidAPS_LOG_1702541012663.log.zip)

[image]
[image]
[image]
[image]
[image]
[image]
[image]

idalioreis, 2023-12-14:
Hi,

Same on me, Pump change insulin add to IOB.

parapenT1sta, 2023-12-19:
@dv1 Can you check it?

MilosKozak, 2023-12-20:
fixed by https://github.com/nightscout/AndroidAPS/commit/405d013d25b462fd903b6472847af38860de1903

closed 2023-12-20 by MilosKozak