nightscout/AndroidAPS #3112: Automation Recurring Time

belevine, 2023-12-14:
AAPS 3.2.0.2

In automations, it is not possible to create a time range for recurring times.  This would be useful to select only specific days of the week to have an action taken during a time range.  Currently, recurring time automations can only be set to occur at a specific time, and time ranges occur on all days. 

This request is to have the day of week selection added to the time range trigger.

TimVroom, 2024-01-26:
At the moment I have a locally edited version that includes this:
Just a simplified version of the recurring time trigger which you can only select the date.
This works great in combination with a time range or any of the other triggers for me.

So my question is:

why the need to include the other already existing triggers into one for recurring purposes? Isn't a single day selector more then enough?

I could make a PR for this, but im not sure about all the inner workings, so not sure it would be accepted. (will link this issue just in case)

belevine, 2024-01-28:
> At the moment I have a locally edited version that includes this: Just a simplified version of the recurring time trigger which you can only select the date. This works great in combination with a time range or any of the other triggers for me.
> 
> So my question is:
> 
> why the need to include the other already existing triggers into one for recurring purposes? Isn't a single day selector more then enough?
> 
> I could make a PR for this if there is any interest, but im not sure about all the inner workings, so not sure it would be accepted. (will link this issue just in case)

Yes, a day selector in combination with a time range is what I am looking to do. I can't figure out how to do that with the current triggers; is it possible?

TimVroom, 2024-01-29:
@belevine Seems like the PR is accepted: so in the upcomming version it should be working.
So with the current stable version it is not possible (as far as i know)

belevine, 2024-01-30:
> @belevine Seems like the PR is accepted: so in the upcomming version it should be working. So with the current 3.2.x version it is not possible (as far as i know)

Thank you!

PR #3202 (2024-01-29): Add day automation
Fix #3112 
A certain day only automation (should be able to replace the recurringTime automation someday).

Copy of and reduced recurring time automation, so it's useable with other automations.