nightscout/AndroidAPS #3117: Alarm and notify of AndroidAPS not bypass limitation set volume system

lostboy86, 2023-12-16:
Last night my rileylink not link with bluetooth, AndroidAPS correctly send notify to phone "Micro not rachable", but compare to xdrip, juggluco, diabhetes app, AndroidAPS don't bypass limitation set volume system or maximize low volume / silent mode.

In my case, i unfortunately set zero volume for my work in the evening for 1 hours, and forget to re-maximize.

This issue appear for all notify from AndroidAPS, i think it can be an important improvement (also to bypass Do Not Disturb mode for example).

Hope i write all clearly, sorry for my bad english.
Thank you for the fantastic support.

  Luca

CaroGo, 2024-02-15:
Same problem...the alert 'pump not reachable' does not come up if phone is silenced.  Had the issue if drained pump battery during night...no sound, so no reaction from my side...no loop no insulin for more than 3 h! 🫠 I would need the possibility that aaps overrides the silent mode like xdrip does!
Thanks in advance! ☺️

MilosKozak, 2024-02-16:
Look for another issues where people asks to respect DND mode

lostboy86, 2024-02-16:
Maybe implement a switch to enable/disable overwrite bypass sound limit
system settings?

MilosKozak, 2026-10-11:
AAPS now has the setting "Override Do Not Disturb mode" (on by default). AAPS alarms like "Pump unreachable" then play on the alarm sound channel, also when the phone is silenced or in Do Not Disturb. Please reopen if it still does not work on a current version.

closed 2026-10-11 by MilosKozak