nightscout/AndroidAPS #3118: Temporary target from AAPSClient

vmichalec, 2023-12-16:
* AAPS Version: 3.2.0.2
* Connection using NSClientV3 over websocket, temp target sync enabled
* NS 15.0.3

Let's assume aaps with SMB disabled with high temp target. We have active temp target of 8 and 10 minutes remaining.
On AAPS side I can just extend this temp target by entering new one without issues below.

On NSClient side I can also set new temp target but there is different behavior:

1. during the sync on AAPS side there can be SMB issued in between those two high temp targets (looks like cancelling current TT, running the loop and maybe give some insulin, setting new TT that is coming).
2. if there is active automation rule to set temp target for example to 5, this one wins and suppress the one coming from remote (such active rule effectively blocks manually set TT from remote).

painbrain81, 2023-12-17:
it's solved with 3.2.0.3 i think

MilosKozak, 2023-12-20:
@vmichalec can you re-test with 3.2.0.3 ?

vmichalec, 2023-12-20:
Yes, I will update tomorrow and check then.

vmichalec, 2023-12-22:
I tested it shortly on 3.2.0.3:
Issue 1) I didn't noticed this one again.
Issue 2) Still there. I have an automation rule from 21:00 for TT of 5.0 and this rule mostly wins over TT from remote.
I'm sending logs to email.

MilosKozak, 2024-01-05:
ad 2) i need screenshots what happened (overview, treatments - TT), exact automation rule, then logs with time of occurence

vmichalec, 2024-01-05:
Ok, fresh case, started after 15:00
- I have created a rule named `TT_DOWN_TO_5` (glucose >= 5.5 triggers TT 5.0 for 30 min)
-  Here is the user activity log on the `AAPS` side
   - 15:03 `TT_DOWN_TO_5` is executed
   - 15:06 I have just set TT on the `AAPS` phone

[image]

- And the TT list on the `AAPS` phone

[image]

- On the `AAPSClient` side
   - 15:10:16 I have set TT 

[image]

- And the TT list on the `AAPSClient` phone

[image]

- Finally we can return to the first/second screenshot of `AAPS` and see what happened after 15:10:16
   - 15:10:22 `TT_DOWN_TO_5` is executed
   - on the TT tab we can see two TT starting at 15:10 and this is the issue, simply said - automation rule TT wins over the TT from the `AAPSClient`

I have also noticed few exceptions like this one in the log:
```
15:10:26.615 [RxCachedThreadScheduler-47] D/CORE: [UserEntryLoggerImpl$log$2.accept():61]: FAILED USER ENTRY: java.lang.NullPointerException: Parameter specified as non-null is null: method app.aaps.database.impl.Converters$ValueWithUnitWrapper.<init>, parameter wrapped [UserEntry(id=0, timestamp=1704386157814, utcOffset=3600000, action=TT, source=NSClient, note=, values=[TherapyEventTTReason(value=HYPOGLYCEMIA), Mgdl(value=144.0), null, Minute(value=60)])]
```
To summarize my config:
- aaps & client: 3.2.0.3
- nightscout 15.0.3
- there are two aaps-clients with full rights connected to this nightscout (subjects `vlado-full`, `betuta-full` )
- I just wonder why there are so many logs rows with `"modifiedBy":"vlado-full"` or even `"modifiedBy":"betuta-full"` when I set TT from client phone `vlado-full` only once and client phone `betuta-full` did not changed anything. Please only check if works as it should.
- I'm sending logs to email.

MilosKozak, 2024-02-11:
put the log here pls. I cannot find it in email

vmichalec, 2024-02-11:
[Issue_3118.zip](https://github.com/nightscout/AndroidAPS/files/14234003/Issue_3118.zip)

closed 2024-11-30 by MilosKozak