nightscout/AndroidAPS #3119: If the OK button is missing. Quite stupid for the kids ;)

elgrupo123, 2023-12-16:
If the OK button is missing. Quite stupid for the kids ;) 3.2.0.3 

Thank you for your hateful work!

[image]

MilosKozak, 2023-12-17:
just scroll down

closed 2023-12-17 by MilosKozak

elgrupo123, 2023-12-17:
well, for us as parents it's self-evident. Unfortunately for an 8 year old it's not. After all, the system is very innovative and should also be usable by small children.

fezulin, 2023-12-18:
I took a closer look at the GUI. Although I was initially unhappy with the change, I have come to the conclusion that the design decision was a good one. 
If you have a small display (like me - Pixel 4), deactivate the checkbox next to the eye at the height of the "Cancel" and "OK" buttons. The display is clearer and the buttons are immediately accessible.
I hope that there are not too many people whose display is too small to show the buttons. 
@MilosKozak: Well done - thank you!

MilosKozak, 2023-12-19:
> well, for us as parents it's self-evident. Unfortunately for an 8 year old it's not. After all, the system is very innovative and should also be usable by small children.

If the patient is 8yo why do you have all details enabled?

elgrupo123, 2023-12-19:
So that he can click away IOB/COB if he has to deliver insulin more often in a row.

MilosKozak, 2023-12-19:
??