nightscout/AndroidAPS #3126: Eros alerts option label typo wrong

lostboy86, 2023-12-18:
There is only a possibility to set alerts relative to 80h for Eros. 

The text in the configuration page is wrong. 

It shuld say 80h not 72h. for alert configurations.

Tested with other users with Eros, it's an error typo on Eros preferences screen

MilosKozak, 2023-12-20:
"Reminder at hours before expiry (72 Hours)" this should be 80?

fredrik-ljung, 2023-12-20:
Yes. It shold be relative 80 Hours. I tested it a couple of days ago.
There is a similar string for dash. With dash you have the possibility to set one alert relative to 80h and one relative to 72h.
Perhaps the description string with 80h for dash can be reused for eros.

MilosKozak, 2023-12-20:
https://github.com/nightscout/AndroidAPS/commit/2c992dfd6824d52a3554df5794cccc7b70c4a037

closed 2023-12-20 by MilosKozak