nightscout/AndroidAPS #3132: [AndroidAPS 3.2.0.3] Suggestion: GUI change bolus wizard

fezulin, 2023-12-20:
I would like to suggest a GUI change for bolus wizard.
The line "carb time" can be included in the table with the other input fields. 
If the details are hidden, the line could still be visible. 
See attached pictures.
I changed the file "dialog_wizard.xml" locally and I like to provide this file. 
(But I also like to avoid the effort to learn the processes about pull requests.)
Please give me a hint how I can share the changed file, if you are interested.

[image]
[image]

MilosKozak, 2023-12-20:
paste it here if you don't know how to create a PR

fezulin, 2023-12-20:
<?xml version="1.0" encoding="utf-8"?>
<ScrollView xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    xmlns:tools="http://schemas.android.com/tools"
    style="@style/StyleDialog"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    android:focusableInTouchMode="true"
    tools:context=".dialogs.InsulinDialog">

    <LinearLayout
        android:layout_width="match_parent"
        android:layout_height="wrap_content"
        android:orientation="vertical">

        <RelativeLayout
            style="@style/StyleDialogHeader"
            android:layout_width="match_parent"
            android:layout_height="wrap_content"
            android:layout_gravity="center">

            <ImageView
                android:layout_width="wrap_content"
                android:layout_height="match_parent"
                android:importantForAccessibility="no"
                app:srcCompat="@drawable/ic_calculator" />

            <TextView
                android:layout_width="match_parent"
                android:layout_height="wrap_content"
                android:layout_centerInParent="true"
                android:layout_gravity="center"
                android:layout_marginStart="10dp"
                android:layout_marginEnd="10dp"
                android:text="@string/boluswizard"
                android:textAlignment="center"
                android:textAppearance="?android:attr/textAppearanceLarge" />

        </RelativeLayout>

        <LinearLayout
            android:layout_width="match_parent"
            android:layout_height="wrap_content"
            android:layout_gravity="center_horizontal"
            android:layout_marginStart="10dp"
            android:layout_marginTop="5dp"
            android:layout_marginEnd="10dp"
            android:background="?attr/dialogTitleBackground"
            android:gravity="center_horizontal"
            android:orientation="horizontal"
            android:paddingTop="5dp"
            android:paddingBottom="5dp">

            <TextView
                android:id="@+id/total"
                android:layout_width="wrap_content"
                android:layout_height="wrap_content"
                android:paddingStart="10dp"
                android:paddingEnd="10dp"
                android:text="2.35U 28g"
                android:textAppearance="?android:attr/textAppearanceLarge"
                tools:ignore="HardcodedText" />

            <TextView
                android:id="@+id/percent_used"
                android:layout_width="wrap_content"
                android:layout_height="wrap_content"
                android:layout_marginStart="10dp"
                android:paddingStart="10dp"
                android:paddingEnd="10dp"
[... 598 lines cut ...]
            android:layout_height="50dp"
            android:layout_gravity="center_vertical"
            android:background="@android:color/transparent"
            android:orientation="horizontal"
            android:paddingStart="5dp"
            android:paddingEnd="5dp">

            <CheckBox
                android:id="@+id/calculation_checkbox"
                android:layout_width="wrap_content"
                android:layout_height="fill_parent"
                android:layout_gravity="center_vertical"
                android:checked="false"
                android:contentDescription="@string/show_calculation"
                android:drawableEnd="@drawable/ic_visibility" />

            <include
                android:id="@+id/okcancel"
                layout="@layout/okcancel" />

        </LinearLayout>

    </LinearLayout>

</ScrollView>

fezulin, 2023-12-20:
Hm - in my last comment the xml-header is not displayed.
I did not change it. Here the header again:

<?xml version="1.0" encoding="utf-8"?>
<ScrollView xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    xmlns:tools="http://schemas.android.com/tools"
    style="@style/StyleDialog"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    android:focusableInTouchMode="true"
    tools:context=".dialogs.InsulinDialog">

fezulin, 2023-12-20:
WTF? The header cannot be displayed here? 
However - it is unchanged anyway.

MilosKozak, 2023-12-20:
https://github.com/nightscout/AndroidAPS/commit/8ab0b697d9bcb68815c2de72b5571ea7a0c524ea

closed 2023-12-20 by MilosKozak