nightscout/AndroidAPS #3136: AAPS displaying 1 day old sensor as 11 days old.

jgslade, 2023-12-21:
I upgraded to 3.2.0.3 today and am running the latest version of xdrip. I got the sensor start time in AAPS now, yay, but it is showing my sensor at 11 days old instead of 1.

I am using the Dexcom G7 sensor.

Screenshot of AAPS
[image]

Screenshot of xdrip
[image]

logs
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/13734643/AndroidAPS.log)

MilosKozak, 2023-12-21:
what is the time of start of sensor?

jgslade, 2023-12-21:
I checked in the systems status on xdrip and found this: 

[image]

[image]

So it seems that this is an xdrip bug.