nightscout/AndroidAPS #3137: AAPS Wear, latest dev commit#8836f90 : Ongoing notification that can not be cleared.

vanelsberg, 2023-12-22:
**AndroidAPS 3.3.0-dev Wear application**
Build: https://github.com/nightscout/AndroidAPS/commit/8836f90a59e29de9f5a6f47a8e546d156c2f2b67
Flavor: Fullrelease

**Problem description:**

Latest dev commit [8836f90](https://github.com/nightscout/AndroidAPS/commit/8836f90a59e29de9f5a6f47a8e546d156c2f2b67) introduces a problem for my Samsung Watch 4 running the AAPS watchface (both stock or custom:

The watch now continuesly shows an "ongoing notification" that can not be cleared.

[image]

vanelsberg, 2023-12-22:
Additional note:
Looks like this problem is only for bolus notifications. They stay open at 100% - Success (see screenschot)
(Notification not properly closed when bolus finisces?)

Carbs notification for instance are not affected.

vanelsberg, 2023-12-22:
**Additional note:**
@MilosKozak Tested with previous commit f3be1bbfe1fe78aee30d791ae5f5e6dadab579b1
Problem is not in this commit.

MilosKozak, 2023-12-24:
ongoing notification channel is necessary to keep app running on background
you have to hide it https://www.reddit.com/r/tasker/comments/rf7369/protip_disable_specific_notification_channels_on/

vanelsberg, 2023-12-24:
> ongoing notification channel is necessary to keep app running on background you have to hide it https://www.reddit.com/r/tasker/comments/rf7369/protip_disable_specific_notification_channels_on/

Nope, does not solve the problem... :-(
From developer mode on the watch I can disable some of the notification types, but as soon as there is a (bolus) notification it  is a "ongoing notification" which can not be cleared. I can disable "all", but then I do not get any notifications at all :-|

**My conclusion:**
1) User being unable to clear (any) notification on the watch UI is unwanted behavior.
2) Hiding only works when disabling _all AAPS notifications_ on the watch, which would disable important functionality.
3) It is rather odd to expect AAPS users to go into developer mode to hide/prevent such unexpected and unwanted behavior?

**Proposed solution:**
Make activating the ongoing channel a user configurable option:
The ongoing notification channel is unexpected and unwanted behavior and only necessary (as a work around?) for users running non-AAPS watchfaces.

@MilosKozak Let me know if you'd like this prosed solution moved to a separate issue?

vanelsberg, 2023-12-25:
@MilosKozak Did some local testing enabling the [Wear: create notification channel](https://github.com/nightscout/AndroidAPS/commit/041d0f0b7bd43042dbf201f442c1a48433eb22cd) as a user option through the wear app settings. 

Code is still QAD, but could to a PR for dev's to test?

closed 2023-12-26 by MilosKozak

vanelsberg, 2023-12-27:
Confirming #3141solved this problem!
Tnx.

PR #3141 (2023-12-26): Wear: Foreground Service Notification ID
This PR introduces a unique notification id for the foreground service in the wear app. It was using 1, which is already assigned to the bolus progress notifications. Probably fixes #3137.