nightscout/AndroidAPS #3140: Red numbers when sensor has life left

robertrub, 2023-12-25:
Dev 3.3.0. I reset sensor life to 14 days but I still have a red 8 days on the UI.

[image]

[image]

[image]

[image]

robertrub, 2023-12-25:
Deleting the CGM insertion in the treatments and entering it again didn't solve the problem. 8 days sensor life is still red.

mick711, 2023-12-25:
With last dev 3 days réd.

mordechi-david, 2023-12-26:
Same to me too

vanelsberg, 2024-01-02:
For me the problem was fixed through (https://github.com/nightscout/AndroidAPS/commit/45d93f0a7c95fe8a3d91f03d4b0a9748ea77c97c)

closed 2024-01-02 by MilosKozak