nightscout/AndroidAPS #3142: AAPS 3.3.0-dev Report insulin without injection not captured

mordechi-david, 2023-12-26:
I reported about insulin that I gave in a syringe so that it would take into account and stop the basal, and it was not absorbed and was not taken into account, I went back to 3.2.0.3 and it was absorbed, I went back to 3.3.0 and again it was not absorbed so this is a problem in this version
Attach logs it happened for the first time at 11:54 (maybe a minute before or after)
And a second time at 12:09
Thanks for all the improvements and hard work
[AndroidAPS (1).log](https://github.com/nightscout/AndroidAPS/files/13770711/AndroidAPS.1.log)
[AndroidAPS._2023-12-19_23-53-23_.4.zip](https://github.com/nightscout/AndroidAPS/files/13770712/AndroidAPS._2023-12-19_23-53-23_.4.zip)

vanelsberg, 2023-12-27:
Confirming!

vanelsberg, 2024-01-02:
Solved for latest 3.2.0-dev

closed 2024-01-02 by MilosKozak