nightscout/AndroidAPS #3143: Wear : Data layer foreground notification

robertrub, 2023-12-27:
Latest Dev on phone and wear.

[image]

I keep getting the Data layer foreground notification over and over.
[image]

 Can't erase it.

[image]

The only option is open the notification settings and stop the notifications but it then asks me if I want to stop all notifications from AAPS... 

[image]

Was working great before. I don't know why this commit was added but it made things worst.

Logs. 

[image]

robertrub, 2023-12-27:
Turning OFF the notification turned OFF ALL notifications on watch from AAPS.

https://github.com/nightscout/AndroidAPS/commit/d75c4651f6e41b44486c8b13500d6b5972ae406a  Is not working as it should.

robertrub, 2023-12-27:
Updated to the latest Dev on phone and watch tonight and the problem is still there.

[image]

vanelsberg, 2023-12-29:
Confirming:
See also https://github.com/nightscout/AndroidAPS/issues/3137 (AAPS Wear: Ongoing bolus notification that can not be cleared)

MilosKozak, 2024-01-02:
on android 10+ ongoing notification must be in place if you want to have service running on background

robertrub, 2024-01-03:
Till now, the notifications were working great . Now we're stuck with a notification that we can't erase. 

The fix made things worst (at least for some (or many ?) users).

MilosKozak, 2024-01-03:
in 3.2.0.3 service was suspended without active watchface (on background). if you add notification android allows have it running on backgroung

vanelsberg, 2024-01-03:
> in 3.2.0.3 service was suspended without active watchface (on background) and complications were not working. if you add notification android allows service running on background. the price for it is permanent notification

@MilosKozak @robertrub How about making this notification feature optional (read: configurable on/off)?
I could do a PR for this...

robertrub, 2024-01-03:
Totally vote for it being an option.

MilosKozak, 2024-01-03:
totally against more options :)

robertrub, 2024-01-03:
You're the boss ;)

How many people use the AAPS watchface and how many use others ?

I still believe that more will be bothered with the change than happy to gave it.

Of course, you can ignore the _stuck notification_ on the watch but I'll be the one to tell them to ignore it when they'll complain about it on Discord... :(

TebbeUbben, 2024-01-12:
@MilosKozak As far as I know, using an ongoing notification is not required, although the system might enforce that depending on the Android version. It might also be worth an attempt to use PRIORITY_MIN. These notifications did not appear on my watch at all and I got a notification by the system instead that AndroidAPS was running in the background (which can be dismissed).

Anyways, on my Pixel Watch 2 I'm able to disable the notification channel in the system settings. Perhaps that setting is hidden on watches from other manufacturers. Manually launching the intent might work (see https://stackoverflow.com/questions/32366649/any-way-to-link-to-the-android-notification-settings-for-my-app). There's probably also a way to manage notifications via ADB.

vanelsberg, 2024-01-13:
> ...] Perhaps that setting is hidden on watches from other manufacturers [...

With current dev-builds, on the Samsung Galaxy Watch 4, I can hide the ongoing notification through developer settings:
`App Notifications -> All watch apps -> AAPS -> Data layer Foreground Service Channel: OFF`

robertrub, 2024-01-14:
No, I don't have that option. Samsung Android 14. Both phone and watch are set to developer mode.

Am I searching in the wrong place ?

[image]

vanelsberg, 2024-01-14:
> No, I don't have that option...

Wrong place. Check developer options menu _on Watch_:
1. Scroll all to way down to App Notifications
2. Select: All watch apps -> AAPS -> Data layer Foreground Service Channel
3. Switch ir OFF

robertrub, 2024-01-14:
OK, thanks. Found it and deactivated it with succes. 

Not the best option/solution for an "average" user though.

Robby

olorinmaia, 2024-01-15:
@MilosKozak Here are my thoughts on the matter. I don't think we want to solve a problem for a few users, and create a new problem that will concern all. As it stands now the data layer foreground notification is an annoyance that most users will have issues with. It vibrates often without any reason, and if you enact anything from the tiles it also comes up first, before what you usually see.

I've been trying to alter the code to make it less annoying, it seems that the following helps, keeping it from blocking wanted notifications when enacting from tiles and vibrating alot less. It still isn't possible to remove tho, and I find that troubling, as it very often adds a dot to the watch screen making it look like you have a missed notification. Which is usually the datalayer foreground notification :/
[image]
[image]

I have tested a few days without commit 041d0f0b to pursue @TebbeUbben claim that a notification channel is not necessary, and so far, using stock samsung galaxy watch watchface I'm still able to enact from all commands from every tile. So maybe a persistent channel isn't necessary? Would be nice if more could test this, so we don't add stuff that isn't really necessary and just is in the way of notifications you really want to see.

vanelsberg, 2024-01-15:
> I have tested a few days without commit https://github.com/nightscout/AndroidAPS/commit/041d0f0b7bd43042dbf201f442c1a48433eb22cd

Originally had the idea disable the change based on a watch config option setting.

However, when I disable the this commit in current dev by commenting out the changes (code from the first screenshot) somehow leaves the wear app unstable in that what looks like the watchface restarting about every minute or so?
Did not get to load this into the debugger (need to figure out how to do that ... ) to see what it is doing...

olorinmaia, 2024-01-15:
> > I have tested a few days without commit [041d0f0](https://github.com/nightscout/AndroidAPS/commit/041d0f0b7bd43042dbf201f442c1a48433eb22cd)
> 
> Originally had the idea disable the change based on a watch config option setting.
> 
> However, when I disable this commit in current dev by commenting out the changes (see code from the first screenshot) somehow this leaves the wear app unstable in that what looks like the watchface restarting about every minute or so? Did not get to load this into the debugger (need to figure out how to do that ... ) to see what it is doing...

I just deleted the lines in that commit, shouldn't matter if delete or comment out I guess, but I'm not experiencing unstable wear app. Neither with our without AAPS watchface.

MilosKozak, 2024-01-17:
the reason why I added notification channel is that in logs were a message that communication service was killed because it didn't call startforeground. That disappeared after mentioned change. My thoughts is that complications cannot work properly without this service.
The change what started this is targeting API 29 on watch in last master version

vanelsberg, 2024-01-17:
> the reason why I added notification channel ...

Thanks for clarifying, which would explain.
Seems solving this is not as straightforward as I anticipated.

olorinmaia, 2024-01-22:
Please test [8cfa312](https://github.com/nightscout/AndroidAPS/commit/8cfa312c087360334d426b2ee92a2682dcf79ebb) and see if it's better now. At least for me it works great. (Samsung Galaxy watch 4)

closed 2024-01-24 by MilosKozak