nightscout/AndroidAPS #3148: Can't get past objective 1 in fresh AAPS install with NSClientV3

olorinmaia, 2023-12-30:
AAPS 3.2.0.3
Build: bb1a276078-2023.12.17
Nightscout version: 15.0.2

I've seen some new users having issues with this, so I have recreated the problem. Steps:
1) Do a fresh install of AAPS
2) configure NSclientv3
3) enable all uploads etc to NS

Alltho BG is uploaded to NS, the objective regarding this doesnt get completed.

Workaround is to select NSClientV1, then objective gets completed.

Logs:
[AndroidAPS_LOG_1703899762554.log.zip](https://github.com/nightscout/AndroidAPS/files/13796768/AndroidAPS_LOG_1703899762554.log.zip)

Screens:
[image]
[image]
[image]

emna2305, 2024-01-13:
i got the same problem how i can fix it

olorinmaia, 2024-01-13:
Switch to NSClientv1 and see that objective pass then switch back

olorinmaia, 2024-01-17:
Fixed by https://github.com/nightscout/AndroidAPS/commit/172075ebf80a929004dfd8be5d8863beda42712b

closed 2024-01-17 by olorinmaia