nightscout/AndroidAPS #3149: Insulin suspended notification

robertrub, 2023-12-30:
Last Dev

[image]

A bolus is being given but a notification says insulin suspended...

[image]

[image]

Maybe related, there was a temporary 50% profile coming to it's end. Don't know how long it still had.

Logs. Look around 9h19

[image]

MilosKozak, 2024-01-01:
notification was genereated 23:47

robertrub, 2024-01-02:
Shouldn't the notification be deleted by itself when the insulin started being given again ? It was probably a temporary message during a profile change by automation as at that time, I was surely sleeping.

MilosKozak, 2024-01-02:
this message is not confirmed automatically

robertrub, 2024-01-03:
Ok. Looks like something changed in the DASH driver. This message was almost never stuck (or maybe shown ??) till very recently. I only got it when pump was far from phone. 

Now, it seems to happen once a day. When I see the notification and check the DASH tab, I need to reactivate insulin distribution (but not sure if I need to do it systematically...)

Will go on observing to give more precise details.

I'm not sure if this didn't start since @Philoul corrected an untranslated message to translatable (message was _uncertain_ ). Can't see why/if it changed anything... Wild guess only ;)

closed 2024-01-12 by MilosKozak