nightscout/AndroidAPS #3151: No more logs

robertrub, 2023-12-30:
My last logs are dated Dec 25th. Something in the code stopped working when you installed unfinished mode.

This is my last Dev version but it stopped working before this one

[image]

robertrub, 2023-12-30:
I found the screenshot of the version I had installed on the 25th in the waste basket...

[image]

MilosKozak, 2024-01-01:
with new libs version it's not possible to keep logs in `/AAPS/logs` directory and they are back in default location `/Android/data/info.nightscout.androidaps/files`

closed 2024-01-01 by MilosKozak